openai/mujoco-py #282: Meaning of the observation list

theobdt, 2018-08-30:
Hi, 
Is there a way of knowing the "meaning" of the elements from the observation list ? There is good documentation for the basic environments : 
[https://github.com/openai/gym/wiki/CartPole-v0](url)
But I can't find something similar for the mujocos envs. Thank you very much.

BrunoBSM, 2018-09-01:
I agree, the Mujoco envs don't have such good doccumentations. If you've played around with mujoco-py you may recognize some names within the `_get_obs()` function (like in the ant environment https://github.com/openai/gym/blob/master/gym/envs/mujoco/ant.py ), like `qpos` and `qvel` which correspond to cartesian positions, rotation quartenions and velocities for joints. On other environments there are some names I do not recognize like `cfrc`. In order to find out what they are I recommend looking at the MuJoCo doccumentation http://www.mujoco.org/book/ and playing around with mujoco-py's `sim.data` and `sim.model` with a known simmulation you know how many and where are its joints. By using `variables = [i for i in dir(sim.model)]` and then printing `variables` you can see how many variables and functions are inside each one for you to read or write. Hope this helps, and I hope someone comes with a much simpler way.

machinaut, 2018-10-17:
@theobdt this package is the python bindings to the mujoco library, so really just focused on physics simulation.

Reinforcement Learning environments (and their observations, etc) mostly live in different packages.

If this question is about an environment, it'd be better to ask that on the repo for that environment.

Similarly, MuJoCo is documented in its own book here: http://www.mujoco.org/book/index.html and if you have questions about the mujoco library, there is a forum here http://www.mujoco.org/forum/index.php

closed 2018-10-17 by machinaut

Krishnendu1984, 2022-08-24:
Can anyone tell me how to extract location or coordinates  of ground reaction  contact force in mujoco