openai/mujoco-py #284: command 'gcc' failed with exit status 1

inspirepassion, 2018-09-01:
I followed the instruction (https://github.com/openai/mujoco-py/), and everything is good until I `import mujoco_py`, could anyone help with that? Thank you in advance.

 
>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/jason/.conda/envs/gpu/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/jason/.mujoco/mujoco-py/mujoco_py -I/home/jason/.mujoco/mjpro150/include -I/home/jason/.conda/envs/gpu/lib/python3.6/site-packages/numpy/core/include -I/home/jason/.conda/envs/gpu/include/python3.6m -c /home/jason/.mujoco/mujoco-py/mujoco_py/cymj.c -o /home/jason/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/jason/.mujoco/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/jason/.conda/envs/gpu/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Imujoco_py -I/home/jason/.mujoco/mujoco-py/mujoco_py -I/home/jason/.mujoco/mjpro150/include -I/home/jason/.conda/envs/gpu/lib/python3.6/site-packages/numpy/core/include -I/home/jason/.conda/envs/gpu/include/python3.6m -c /home/jason/.mujoco/mujoco-py/mujoco_py/gl/osmesashim.c -o /home/jason/.mujoco/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.56_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/jason/.mujoco/mujoco-py/mujoco_py/gl/osmesashim.o -fopenmp -w
/home/jason/.mujoco/mujoco-py/mujoco_py/gl/osmesashim.c:1:10: fatal error: GL/osmesa.h: No such file or directory
 #include <GL/osmesa.h>
          ^~~~~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/home/jason/.conda/envs/gpu/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/home/jason/.conda/envs/gpu/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/jason/.conda/envs/gpu/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/jason/.conda/envs/gpu/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

BrunoBSM, 2018-09-01:
Well, for me, when I tried the exact instructions I wasn't able to make it work either. What solved for me was installing the `dependencies.txt` and `dependencies.dev.txt` before installing mujoco-py

inspirepassion, 2018-09-01:
@BrunoBSM Could you please tell me how to install that two txt files? I’m a newbie to the system?

BrunoBSM, 2018-09-01:
Sure thing, I had the name wrong before, you can simply run 
`sudo pip install -r requirements.txt`
and then 
`sudo pip install -r requirements.dev.txt`
and make sure you have the python version right, I had some real problems with that also

inspirepassion, 2018-09-01:
@BrunoBSM Thanks for the reply, and apologize to this late reply, as my ubuntu crushed yesterday when I tried different things. I just reinstalled the whole system, and set up all the previous preparation. 

Now I tried the command, it reports this error:

(gpu36) zhixun@zhixun-MS-7B48:~$ pip install -r requirements.txt
Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'
You are using pip version 10.0.1, however version 18.0 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

Do you by any change know about his? Thank you in advance

Jerryxiaoyu, 2018-09-02:
same problem...

karamarieliu, 2018-09-11:
same error. tried the installation of requirements first, didn't work

mayou36, 2019-01-04:
@LJ147 worked for me! I had the same problem and the installation with Master seems to work...

YuYangguang, 2019-01-23:
please type in the shell as
`sudo apt-get install libosmesa6-dev`
This works for me. Hope it can help you

CSYing95, 2019-04-07:
@LJ147 worked for me too! Thanks.

CSYing95, 2019-04-07:
@LJ147 It seems to work when installing it with bash. However, when I import mujoco_py pkg in python, it starts compiling and fails with 'gcc' error

shong-unist, 2019-10-15:
In my case, missing the package, GL/glew.h, causes this problem. Thus, I installed the 'libglew-dev' package though `sudo apt install libflew-dev`. (refer to [this](https://stackoverflow.com/questions/15852417/compiling-opengl-program-gl-glew-h-missing))

levifussell, 2020-03-20:
I had a slightly different error which was as above but with "fatal error: limits.h: No such file or directory" which I got to after moving beyond the above error by following @YuYangguang, above. To fix this next error (which occurred when trying to import in python interpreter) I did:

(1) Install mujoco-py from source.
(2) install ```limits.h``` via library ```apt-get install libc-dev``` as per [here](https://stackoverflow.com/questions/30624829/no-such-file-or-directory-limits-h-when-installing-pillow-on-alpine-linux)
(3) install patchelf as per ```conda install patchelf```

MonaHe123, 2020-04-17:
You can try the way bellow,it works for me:
sudo apt-get install -y python-numpy python-dev cmake zlib1g-dev libjpeg-dev xvfb libav-tools xorg-dev python-opengl libboost-all-dev libsdl2-dev swig

XueminLiu111, 2020-07-31:
I tried everything above but still failed. Can anyone help me?

mbilalai, 2021-09-08:
@levifussell You're a lifesaver man!

DailinH, 2021-12-21:
> 

Had the same error when installing on ubuntu 20.04. I was able to get over this issue by:
`sudo apt-get install libglew-dev`
`sudo apt install libopengl0 -y`

atonkamanda, 2022-01-22:
> I had a slightly different error which was as above but with "fatal error: limits.h: No such file or directory" which I got to after moving beyond the above error by following @YuYangguang, above. [...]

Installing patchelf fixed it for me (Ubuntu 20.04) thank you very much !

molumitu, 2022-03-27:
@atonkamanda It also works for me (Ubuntu 20.04)

Alipira, 2022-05-13:
for the "GCC" error, it is because of the C compiler ` sudo apt-get install gcc ` and then `sudo apt-get install build-essential` . for me it works

jingxuxie, 2022-10-23:
This works for me on Ubuntu 22.04. Thank you!

BUAAZhangHaonan, 2023-10-25:
> for the "GCC" error, it is because of the C compiler `sudo apt-get install gcc` and then `sudo apt-get install build-essential` . for me it works

Very useful guy!!!It works for me on Ubuntu 23

TianyuCodings, 2024-03-09:
libglfw-dev
libglew-dev
libosmesa6-dev
gcc

Install them by sudo, then it works for me.

MoonOutCloudBack, 2025-01-06:
Change mujoco-py to version 2.0.2.5 seems to save me!

```bash
cd mujoco-py
git checkout v2.0.2.5
pip install -e .
```