openai/mujoco-py #285: Win 10 pip3 install -U 'mujoco-py<1.50.2,>=1.50.1' The system cannot find the file specified

inspirepassion, 2018-09-04:
I am tightly following the instruction to install Mujoco, but when I do 

pip3 install -U 'mujoco-py<1.50.2,>=1.50.1'

The system cannot find the file specified.

Does anyone know how to solve it?

Btw, I also try using the same instruction on Linux 18.04, but some packages dependencies issue going on, please refer this [link](https://askubuntu.com/questions/1071856/ubuntu-18-04-libosmesa6-depends-libglapi-mesa-18-0-0rc5-1ubuntu1-but-18-0-5) to take a look, because I hope developers could help make this work: https://askubuntu.com/questions/1071856/ubuntu-18-04-libosmesa6-depends-libglapi-mesa-18-0-0rc5-1ubuntu1-but-18-0-5

machinaut, 2018-10-17:
What system are you on?  Those specs should match https://pypi.org/project/mujoco-py/

Also, what version of python and pip3 are you using?

If you can't find the package, if you check out this git repo and install the python package from source, does that work?

SSS135, 2018-11-30:
Had same problem on Windows 10 x64. Solved it by replacing single quotes with double.
`python -m pip install -U "mujoco-py<1.50.2,>=1.50.1"`