openai/mujoco-py #289: Mujoco-pu

phaniram-sayapaneni, 2018-09-12:
I'm a newbie and I'm having the following issues:
1. Couldn't use mujoco, since whenever I import it, it is throwing me an error
2. Tried gcc-6,7,8 nothing worked

My env:
python 3.6
Mac

Help me understand the problem, I will try to fix myself. Thanks!

machinaut, 2018-10-17:
Can you provide the error?  Otherwise I do not know how to debug this.

closed 2018-10-17 by machinaut

phaniram-sayapaneni, 2018-10-19:
I think, it happened to me because I had no license.
Is it true that I can't use Mujoco-py for free for more than 30 days? If so, can open make it free for all?

machinaut, 2018-10-19:
Trial licenses are available for a fixed period.

Personal student licenses are free as well, please see this page for more information https://www.roboti.us/license.html

The mujoco forum also has a sub-forum for discussing licenses.  http://www.mujoco.org/forum/index.php

Other than that, I can say we are aware of this problem, and how it limits folks from running some of the gym benchmarks.

I would personally be excited to review pull requests that run the 3d physics gym environments with a different engine.