openai/mujoco-py #291: [help wanted]errors occured when import mujoco_py

shaneshbi, 2018-09-17:
**the problems occured as follows**>>> import mujoco_py
Compiling /usr/local/lib/python3.5/dist-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /usr/local/lib/python3.5/dist-packages/mujoco_py/cymj.pyx
warning: /usr/local/lib/python3.5/dist-packages/mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
warning: /usr/local/lib/python3.5/dist-packages/mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared
/ModuleNode.py", line 400, in generate_c_code
    f = open_new_file(result.c_file)
  File "/usr/local/lib/python3.5/dist-packages/Cython/Utils.py", line 57, in open_new_file
    os.unlink(path)
**PermissionError**: [Errno 13] Permission denied: '/usr/local/lib/python3.5/dist-packages/mujoco_py/cymj.c'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py/builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py/builder.py", line 90, in load_cython_ext
    cext_so_path = builder.build()
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py/builder.py", line 202, in build
    built_so_file_path = self._build_impl()
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py/builder.py", line 274, in _build_impl
    so_file_path = super()._build_impl()
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py/builder.py", line 215, in _build_impl
    dist.ext_modules = cythonize([self.extension])
  File "/usr/local/lib/python3.5/dist-packages/Cython/Build/Dependencies.py", line 1026, in cythonize
    cythonize_one(*args)
  File "/usr/local/lib/python3.5/dist-packages/Cython/Build/Dependencies.py", line 1146, in cythonize_one
    raise CompileError(None, pyx_file)
**Cython.Compiler.Errors.CompileError**: /usr/local/lib/python3.5/dist-packages/mujoco_py/cymj.pyx

i tried to chmod 'cymi.c',but it doesn't work.

machinaut, 2018-10-17:
It looks like the permissions on your compiled files are strange.

What happens when you uninstall and reinstall without cache?

```
pip uninstall mujoco-py
pip install -U --no-cache --no-cache-dir mujoco-py
```

rohansingh42, 2019-07-13:
Facing same issue. Any solutions?

oluwayetty, 2019-11-19:
@rohansingh42 @shaneshbi did you get any solution to this yet? I'm facing the same

cross32768, 2019-12-01:
How about to use "pip install -e ." instead of "python setup.py install" for install? 
This solves a similar problem in my case.

jamesheald, 2019-12-25:
I received the following error when trying to import mujoco_py:

PermissionError: [Errno 13] Permission denied: b'/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/mujoco_py/generated/mujocopy-buildlock'

I solved as follows:

cd /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/mujoco_py
sudo chmod -R 777 ./ (change permission of all files in all subfolders recursively)

yuvalzaidel, 2020-10-14:
Facing the same issue. Any solutions?

DogWY, 2023-09-28:
Hello, my laptop is Macbook Pro with M1Pro, and I ever faced the same issue.I referred to this blog(https://blog.csdn.net/CCCDeric/article/details/131788795) and successfully solved this problem. This is a blog with chinese, It points out that you can try to run this code: `pip install Cython==3.0.0a10` to change your version of Cpython.