openai/mujoco-py #295: alternate install location and discover_mujoco

ysagon, 2018-09-21:
Hello,

I'm trying to install for one of our user mujoco-py. Mujoco itself is installed here:

    /.mujoco/mjpro150/bin/

When I try to load mujoco from Python:

 ```
$ python 
Python 3.6.6 (default, Jun 28 2018, 04:42:43) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/env/lib/python3.6/site-packages/mujoco_py/__init__.py", line 1, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/env/lib/python3.6/site-packages/mujoco_py/builder.py", line 468, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/env/lib/python3.6/site-packages/mujoco_py/builder.py", line 68, in load_cython_ext
    _ensure_set_env_var("LD_LIBRARY_PATH", lib_path)
  File "/env/lib/python3.6/site-packages/mujoco_py/builder.py", line 101, in _ensure_set_env_var
    var_name, var_name, lib_path))
Exception: 
Missing path to your environment variable. 
Current values LD_LIBRARY_PATH=/.mujoco/mjpro150/bin::/.singularity.d/libs
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sagon/.mujoco/mjpro150/bin
```
Is this corect? It seems that the function `discover_mujoco` in this file `/env/lib/python3.6/site-packages/mujoco_py/utils.py` only specify that mujoco is installed in the home directory.    in this file `/env/lib/python3.6/site-packages/mujoco_py/utils.py`

The following variables are set:
```
MUJOCO_PY_MJPRO_PATH=/.mujoco/mjpro150
MUJOCO_PY_MUJOCO_PATH=/.mujoco
```
But it seems they have no effect.

Best

machinaut, 2018-10-17:
Seems like this could be a bug.  What is `$HOME` set to in this case?

closed 2018-10-18 by machinaut

GoingMyWay, 2019-05-25:
@ysagon  As the output said, you should export the path.

`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sagon/.mujoco/mjpro150/bin`

HareshMiriyala, 2020-04-11:
does not work even after setting the LD_LIBRARY_PATH to the bin directory. :/

PR #308 (2018-10-18): set mujoco location with environment variables
Fixes #295
commit 426fea32 (2018-10-18): set mujoco location with environment variables (#308)

Fixes #295