openai/mujoco-py #298: Fatal error - file name too long

iandanforth, 2018-09-28:
```
c1: fatal error C1081: 'C:\Users\Ian\Miniconda3\envs\drlnd\lib\site-packages\mujoco_py-1.50.1.59-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.59_36_windowsextensionbuilder\temp.win-amd64-3.6\Release\Users\Ian\Miniconda3\envs\drlnd\lib\site-packages\mujoco_py-1.50.1.59-py3.6.egg\mujoco_py\cymj.obj': file name too long
```

Given that `conda` is the recommended installation mechanism for `pytorch` I expect other windows users to run into this issue as well. 

There doesn't seem to be anything in that path that isn't mandated by either conda or python.

I'm blocked on this issue so any suggestions are welcome!

iandanforth, 2018-09-28:
I ended up copying a previously compiled version I had outside the conda env into the env dir and then updating the easy-install.pth file that conda maintains to point to the mujoco-py .egg dir.

machinaut, 2018-10-17:
Pull requests welcome for this, but otherwise not sure how to fix this.

Possible that you could hack the internals of cython or the python builder to make it generate different paths, but this looks like a side effect of how those are handled in windows.

machinaut, 2018-10-17:
Tentatively closing for now, but if other users run into this, please reopen the issue

closed 2018-10-17 by machinaut

rebcabin, 2019-04-09:
I ran into this by following the install instructions verbatim. I don't have a previously compiled version, so I am blocked.

c1: fatal error C1081: 'C:\Users\bbeckman\AppData\Local\Programs\Python\Python37\lib\site-packages\mujoco_py-2.0.2.0-py3.7.egg\mujoco_py\generated\_pyxbld_2.0.2.0_37_windowsextensionbuilder\temp.win-amd64-3.7\Release\Users\bbeckman\AppData\Local\Programs\Python\Python37\lib\site-packages\mujoco_py-2.0.2.0-py3.7.egg\mujoco_py\cymj.obj': file name too long

airqualityanthony, 2021-03-11:
@rebcabin I'm having the same issue here, conda install mujoco-py isn't available.

I can see that the path length is just too big for it to compile. Not sure how I'd change it.

moonblue333, 2021-03-21:
still a bug, file name too long.

lyuyan9527, 2021-04-01:
Here is my tricky solution to this problem 
I am installing mujoco_py 2.0.2 on my win10 in Anaconda, the problem is when building the wheel, the temp directory created by mujoco is too long, for which I can't understand both why there is a duplicate in the path generated by the building process and why windows limit the length of a directoy to 260.
Here is my solution:
1. download the source code of mujoco directly from github and unzip it
2. cd to the souce code folder, activate your conda environment and run "python setup.py install", this will install the required files to your conda environment
3. after step2, when you try to import mujoco_py in python, python will do the builing automatically, which will still cause a 'file name too long' problem, but if you check the path it generates here, it no longer exceeds 260 too much, which is around 270 in my case
4. Here is the tricky part, I changed my anaconda path to just one letter, which will shorten the path directly. For example, I changed my conda path from "D:\ProgramData\Anaconda3" to "D:\P\A", also, I changed my environment's name to a single letter "c" as well(conda activate c)
5. after step4, remember to change the Path variables in environment variables to make sure your conda still work properly
6. then just activate your conda environment and start a python console, then just "import mujoco_py" and let the pyhon build
7. remember to change your conda directory and Path variables back. I was able to run the humanoid example and more time needed to check if there is other problem with my method
Hope this method works well for those who have the same problem with me

secondsensei, 2021-04-09:
Issue is persisting

xiaobanni, 2021-04-13:
I also encountered this problem.

pcheng2, 2021-05-26:
I was saved by this repo, please try it if it could solve your problem, gl :)

csyhy1986, 2021-09-15:
> Here is my tricky solution to this problem
> I am installing mujoco_py 2.0.2 on my win10 in Anaconda, the problem is when building the wheel, the temp directory created by mujoco is too long, for [...]

I confront with the same problem and try to solve it following your instruction. However I don't use Anaconda, I use pip to install my packages, what should I do?

lyuyan9527, 2021-09-15:
The main idea here is just to install the package manually and compress the path length of your python folder. Conda here only serves as an environment manager that determines the path of python.exe. So I guess maybe you can just ignore pip and install mujoco and change the path of your python folder manually? Hope this works.

Ianlande, 2024-07-31:
I found another alternative solution for reference：

**Enabling Long Path Support**
Windows 10 and later versions can support long paths exceeding 260 characters. This can be enabled by modifying the registry or group policy.

Registry Modification:
1. Open the Registry Editor (Press Win + R, then type regedit).
2. Navigate to HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem.
3. Look for the key named LongPathsEnabled. If it doesn’t exist, you can create a new DWORD (32-bit) value.
4. Set the value of LongPathsEnabled to 1.

Group Policy Modification:
1. Open the Group Policy Editor (Press Win + R, then type gpedit.msc).
2. Navigate to Computer Configuration > Administrative Templates > System > Filesystem.
3. Find and enable the policy Enable Win32 long paths.

After making these changes, it is recommended to restart the computer to ensure the changes take effect.

Ianlande, 2024-07-31:
> I found another alternative solution for reference：
>
> **Enabling Long Path Support** Windows 10 and later versions can support long paths exceeding 260 characters. This can be enabled by modifying [...]

I found that it didn't always work. Although no "file name too long" error was reported, the code still failed to compile, possibly due to file names being too long. However, this issue wasn't clearly indicated, and I'm not sure what happened...

Finally, I resolved the issue by creating a shorter file path using the following command:
`conda create -p C:\envs\m python=3.10
This created a new environment in a shorter path "C:\envs\m", and I downloaded the Mujoco source code files there. Then, I installed with these commands:
```
pip install -r requirements.txt
pip install -r requirements.dev.txt
python setup.py install
```
This approach worked, and the installation was successful.

EndlessChian, 2025-06-20:
Well, I thought I've cheated the computer, install mujoco>=3.x on my windows desktop won't go (Can't load DLL), however, It seems using the 2.0.x version of the tool is a workaround. **But** note that it's not an easy way either.

First, make sure "cython<3" is recessary, use `pip` to do that.

Then download mujoco200_windows from https://www.roboti.us/download.html, unpack to C:\user\<user_name>\\.mujoco\, rename the folder as `mujoco200`. And remeber get the lincense from https://www.roboti.us/license.html, copy to the same path. You can open simulate.exe in the \bin folder, and test with choicing one model(xml) in \model.

Then add \bin to the environment variables, then goto basedir like `D:\`, we can build from source code to avoid excessively long path `pip download mujoco==2.0.2.5`, rename to \m, `cd m`. OPEN setup.py, add available DLL path to the begin of setup.py:

> `os.add_dll_directory(r"C:\Users\magia\.mujoco\mujoco200\bin")`
> `os.add_dll_directory(r"D:\m\mujoco_py")`

Next change the constraint of `cython>...,<3` in requirement.txt. After all, use `python setup.py build` it should work, move \build\lib\mujoco to site-packages, `python setup.py install` create the installation information final.

**But** you must add these code in your program before import mujoco_py, otherwise your package may be destroyed :-(

> `os.add_dll_directory(r"C:\Users\magia\.mujoco\mujoco200\bin")`
> `os.add_dll_directory(r"D:\python\Lib\site-packages\mujoco_py")`
> `####### Now Test#######`
> `import mujoco_py`
> `mj_path, _ = mujoco_py.utils.discover_mujoco()`
> `xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')`
> `model = mujoco_py.load_model_from_path(xml_path)`
> `sim = mujoco_py.MjSim(model)`
> `print(sim.data.qpos)`
> `# [0.  0.  1.4 1.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0. 0.  0.  0.  0.  0.  0.  0.  0.  0.  0. ]`
> `sim.step()`
> `print(sim.data.qpos)`
> `# exit()`

This is another problem, your dependency libraries **maybe not Compatible with mujoco_py**, but I use it on `gymnasium` well, after some revise.

For example, if I want use models in \site-packages\gymnasium\envs\mujoco, like "InvertedDoublePendulum-v4", which corresponds to inverted_double_pendulum_v4.py, backup the file and rename it, such as inverted_double_pendulum_v4_py.py, replace all usages of mujoco as mujoko_py like `from gymnasium.envs.mujoco import MuJocoPyEnv as MujocoEnv`, then register it in \gymasium\env\\\_\_init\_\_.py

> `register(`
> `    id="PyInvertedDoublePendulum-v4",`
> `    entry_point="gymnasium.envs.mujoco.inverted_double_pendulum_v4_py:InvertedDoublePendulumEnv",`
> `    ...`
> `)`

also the methods not exactly the same sometimes, like there is a `default_camera_config` keyword need to remove in this model.

Now it work well:

> `with set_gym_backend("gymnasium"):`
> `    base_env = GymEnv("PyInvertedDoublePendulum-v4", device=device)`
> `    print(base_env.available_envs)`