openai/mujoco-py #299: fatal error: mjmodel.h: No such file or directory on python 3.6

nikorose87, 2018-10-01:
Hello,
 
This can be an old error but I am confused because it was supposedly fixed, however, importing the `mujoco_py` library, this error come up:

`fatal error: mjmodel.h: No such file or directory`

I am working with CentOS and I have load environments with `gcc` 0.4.9 and 7.1.1 and none of them have worked so far.

I also have tried installing from pip or from source and It did not worked.

I appreciate your help.

ehsanhaghdoust, 2018-10-05:
facing the same issue

machinaut, 2018-10-17:
Did you install mujoco?  where is the header file (part of the installation)?

https://github.com/openai/mujoco-py#install-mujoco

nikorose87, 2018-10-18:
This is the complete line of the error!

`/home/eprietop/.mujoco/mjpro150/mujoco_py/cymj.c:611:10: fatal error: mjmodel.h: No such file or directory
 #include "mjmodel.h"
`

Previously, I have downloaded the `mjpro150` folder and pasted in the same location as the keyfile is. Moreover, I have added the path to the bash, though, It did not work.