openai/mujoco-py #302: How to install mujoco_py in windows7 with anaconda3 and python3.6.5?

LiDuan1111, 2018-10-02:


Liliduan, 2018-10-02:
I follow the readme to setup the mujoco_py,and it shows that it's successful.However,when I try
 to import the mujoco_py,the following error occured:
>>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "f:\软件\安装包\mujoco\mujoco-py-1.50.1.0\mujoco_py\__init__.py", line 1,
 in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, Mujoc
oException
  File "f:\软件\安装包\mujoco\mujoco-py-1.50.1.0\mujoco_py\builder.py", line 283
, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "f:\软件\安装包\mujoco\mujoco-py-1.50.1.0\mujoco_py\builder.py", line 54,
 in load_cython_ext
    cext_so_path = builder.build()
  File "f:\软件\安装包\mujoco\mujoco-py-1.50.1.0\mujoco_py\builder.py", line 131
, in build
    dist.run_commands()
  File "E:\software\anaconda3\lib\distutils\dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "E:\software\anaconda3\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "E:\software\anaconda3\lib\site-packages\Cython\Distutils\old_build_ext.p
y", line 186, in run
    _build_ext.build_ext.run(self)
  File "E:\software\anaconda3\lib\distutils\command\build_ext.py", line 339, in
run
    self.build_extensions()
  File "f:\软件\安装包\mujoco\mujoco-py-1.50.1.0\mujoco_py\builder.py", line 75,
 in build_extensions
    build_ext.build_extensions(self)
  File "E:\software\anaconda3\lib\site-packages\Cython\Distutils\old_build_ext.p
y", line 194, in build_extensions
    self.build_extension(ext)
  File "E:\software\anaconda3\lib\distutils\command\build_ext.py", line 533, in
build_extension
    depends=ext.depends)
  File "E:\software\anaconda3\lib\distutils\_msvccompiler.py", line 345, in comp
ile
    self.initialize()
  File "E:\software\anaconda3\lib\distutils\_msvccompiler.py", line 238, in init
ialize
    vc_env = _get_vc_env(plat_spec)
  File "E:\software\anaconda3\lib\distutils\_msvccompiler.py", line 134, in _get
_vc_env
    raise DistutilsPlatformError("Unable to find vcvarsall.bat")
distutils.errors.DistutilsPlatformError: Unable to find vcvarsall.bat

How should I do to solve it?

closed 2018-10-03 by LiDuan1111

Aafiya-H, 2023-05-27:
Hi @LiDuan1111 ! Have you been able to resolve the issue? If so, what method worked for you?