openai/mujoco-py #303: [Feature request]: Support for Mujoco 2.0

zuoxingdong, 2018-10-02:


gpzhaome, 2018-10-03:
I'm also wondering if Mujoco 2.0 will be supported.

ogroth, 2018-10-12:
+1 on the feature request :)
MuJoCo 2.0 has quite a few useful features (simulation of non-rigid objects, semantic segmentation rendering of individual geoms).
Are there any plans to support the new MuJoCo version or is there even a timeline for a tentative release of mujoco-py 2.0?
Out of curiosity I've tried to wrap the new MuJoCo headers myself by pointing the [wrapper generator](https://github.com/openai/mujoco-py/blob/master/scripts/gen_wrappers.py) to the new files, but the new GUI APIs cause issues since the current wrapping script can't parse C union data structures which are used in the new headers.

machinaut, 2018-10-17:
Right now we don't use it internally, and unfortunately the way bindings were released was very manual and took a lot time to get tests for everything we thought of (but the tests are by no means comprehensive).

Additionally, in the past when we've moved versions, we've found reinforcement learning benchmarks (performance of a trained policy, learning curves of an implementation, etc) change.

However the improved speed is a pretty compelling reason to consider the upgrade.

Inspired individuals are encouraged to look at the `pxd` files, which (although they look like they were autogenerated from headers, it was actually sadly manual) need updating https://github.com/openai/mujoco-py/tree/master/mujoco_py/pxd

I took a quick stab at the update and there were enough changes to header file and struct definitions that all the interfaces are backwards-incompatible (probably why it went from a `1.x` release to a `2.x` release).

The interface changes in the release notes look like it shouldn't be too much work to make a new version -- but since this is incompatible, I think the real question is about whether/when to support this release, and whether to support multiple releases or deprecate the 1.50 series.

release notes https://www.roboti.us/index.html#mujoco200

cc: @jonasschneider

zuoxingdong, 2019-02-04:
@machinaut some news recently regarding upgrading ?

hyc6668378, 2019-02-26:
please update to Mujoco_py 2.0.
I can't wait any more...

kracon7, 2019-03-13:
Also wondering if there's any timeline on support for mujoco2.0.
Really appreciate it.

Ziyu-Ji, 2019-03-13:
Same here. I'm working on non-rigid object. Need it so bad.

MillionIntegrals, 2019-03-14:
You can check out #368, should be merged relatively soon.

MillionIntegrals, 2019-03-19:
This is done now. Just released a version 2.0.2.0a1.

closed 2019-03-19 by MillionIntegrals