openai/mujoco-py #309: This is the complete line of the error!

nikorose87, 2018-10-18:
`/home/eprietop/.mujoco/mjpro150/mujoco_py/cymj.c:611:10: fatal error: mjmodel.h: No such file or directory
 #include "mjmodel.h"
`

Previously, I have downloaded the `mjpro150` folder and pasted in the same location as the keyfile is. Moreover, I have added the path to the bash, though, It did not work.

_Originally posted by @nikorose87 in https://github.com/openai/mujoco-py/issue_comments#issuecomment-431013718_

machinaut, 2018-10-19:
Can you paste the full error, and give more information about your install (python version, mujoco-py version)

nikorose87, 2018-10-19:
Absolutely,

The python version is 3.6.6, build under anaconda 3, the conda list shows the following Mujoco version:

> mujoco-py                 1.50.1.59                 <pip>

the full error is the following:

```
[GCC 4.8.2 20140120 (Red Hat 4.8.2-15)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Compiling /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/cymj.pyx
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:121:4: 'mjtJoint' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:127:4: 'mjtGeom' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:149:4: 'mjtCamLight' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:157:4: 'mjtTexture' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:163:4: 'mjtIntegrator' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:168:4: 'mjtCollision' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:174:4: 'mjtCone' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:179:4: 'mjtJacobian' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:185:4: 'mjtSolver' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:191:4: 'mjtImp' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:198:4: 'mjtRef' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:203:4: 'mjtEq' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:211:4: 'mjtWrap' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:220:4: 'mjtTrn' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:230:4: 'mjtDyn' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:237:4: 'mjtGain' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:242:4: 'mjtBias' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:248:4: 'mjtObj' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:274:4: 'mjtConstraint' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:285:4: 'mjtConstraintState' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:294:4: 'mjtSensor' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:338:4: 'mjtStage' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:345:4: 'mjtDataType' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:121:4: 'mjtJoint' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:127:4: 'mjtGeom' redeclared 
warning: /home/eprietop/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.59-py3.6.egg/mujoco_py/generated/../pxd/mjmodel.pxd:149:4: 'mjtCamLight' redeclared 
[... cut ...]
  File "/home/eprietop/anaconda3/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/home/eprietop/anaconda3/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/eprietop/anaconda3/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/bin/x86_64-pc-linux-gnu-gcc' failed with exit status 1
****
```

machinaut, 2018-10-19:
@nikorose87 this happens when the compiler can't find the mujoco header files.

The default location for these is `$HOME/.mujoco/mjpro150/include` -- can you verify that the mujoco header files are there?

nikorose87, 2018-10-19:
Effectively, I do not have such folder, only the following:

```
build       docs        Makefile     mujoco_py.egg-info    requirements.txt  vendor
dist        examples    MANIFEST.in  README.md             scripts           xmls
Dockerfile  LICENSE.md  mujoco_py    requirements.dev.txt  setup.py
```

machinaut, 2018-10-20:
You might need to follow the mujoco installation instructions here https://github.com/openai/mujoco-py/#install-mujoco

closed 2018-10-21 by nikorose87