openai/mujoco-py #310: fatal error: mjmodel.h: No such file or directory

marctheshark3, 2018-10-19:
Below is the error, i believe others are having this issue but I couldnt find a solution. Also trying this from a Mac. 

Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command "/Users/marctheshark/Documents/MODELING AND SIMULATION/Machine Learning/Final Project/venv/bin/python" -u -c "import setuptools, tokenize;__file__='/private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-j08pcfid/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-record-k64zsvxg/install-record.txt --single-version-externally-managed --compile --install-headers "/Users/marctheshark/Documents/MODELING AND SIMULATION/Machine Learning/Final Project/venv/include/site/python3.7/mujoco-py":
    running install
    running build
    Removing old mujoco_py cext /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-j08pcfid/mujoco-py/mujoco_py/generated/cymj_1.50.1.62_37_macextensionbuilder_37.so
    running build_ext
    building 'mujoco_py.cymj' extension
    /usr/local/bin/gcc-8 -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch x86_64 -g -DONMAC -Imujoco_py -I/private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-j08pcfid/mujoco-py/mujoco_py -I/Users/marctheshark/.mujoco/mjpro150/include -I/Users/marctheshark/Documents/MODELING AND SIMULATION/Machine Learning/Final Project/venv/lib/python3.7/site-packages/numpy/core/include -I/Users/marctheshark/Documents/MODELING AND SIMULATION/Machine Learning/Final Project/venv/include -I/Library/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-j08pcfid/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-j08pcfid/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.62_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-j08pcfid/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-j08pcfid/mujoco-py/mujoco_py/cymj.c:658:10: fatal error: mjmodel.h: No such file or directory
     #include "mjmodel.h"
(venv) Marcs-MacBook-Pro:Final Project marctheshark$ pip3 install mujoco-py --user
Collecting mujoco-py
  Using cached https://files.pythonhosted.org/packages/74/87/02595fadd8ec1a9146745c5c3ea9ff4e3d290fd4216a4a36bc88cba3f522/mujoco-py-1.50.1.62.tar.gz
Requirement already satisfied: glfw>=1.4.0 in ./venv/lib/python3.7/site-packages (from mujoco-py) (1.7.0)
Requirement already satisfied: numpy>=1.11 in ./venv/lib/python3.7/site-packages (from mujoco-py) (1.15.2)
Requirement already satisfied: Cython>=0.27.2 in ./venv/lib/python3.7/site-packages (from mujoco-py) (0.29)
Requirement already satisfied: imageio>=2.1.2 in ./venv/lib/python3.7/site-packages (from mujoco-py) (2.4.1)
Requirement already satisfied: cffi>=1.10 in ./venv/lib/python3.7/site-packages (from mujoco-py) (1.11.5)
Requirement already satisfied: lockfile>=0.12.2 in ./venv/lib/python3.7/site-packages (from mujoco-py) (0.12.2)
Requirement already satisfied: pillow in ./venv/lib/python3.7/site-packages (from imageio>=2.1.2->mujoco-py) (5.3.0)
Requirement already satisfied: pycparser in ./venv/lib/python3.7/site-packages (from cffi>=1.10->mujoco-py) (2.19)
Building wheels for collected packages: mujoco-py
  Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command "/Users/marctheshark/Documents/MODELING AND SIMULATION/Machine Learning/Final Project/venv/bin/python" -u -c "import setuptools, tokenize;__file__='/private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-87ikc8y0/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-wheel-1mcafkgm --python-tag cp37:
  running bdist_wheel
  running build
  Removing old mujoco_py cext /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-87ikc8y0/mujoco-py/mujoco_py/generated/cymj_1.50.1.62_37_macextensionbuilder_37.so
  Compiling /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-87ikc8y0/mujoco-py/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-87ikc8y0/mujoco-py/mujoco_py/cymj.pyx
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:121:4: 'mjtJoint' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:127:4: 'mjtGeom' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:149:4: 'mjtCamLight' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:157:4: 'mjtTexture' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:163:4: 'mjtIntegrator' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:168:4: 'mjtCollision' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:174:4: 'mjtCone' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:179:4: 'mjtJacobian' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:185:4: 'mjtSolver' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:191:4: 'mjtImp' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:198:4: 'mjtRef' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:203:4: 'mjtEq' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:211:4: 'mjtWrap' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:220:4: 'mjtTrn' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:230:4: 'mjtDyn' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:237:4: 'mjtGain' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:242:4: 'mjtBias' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:248:4: 'mjtObj' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:274:4: 'mjtConstraint' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:285:4: 'mjtConstraintState' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:294:4: 'mjtSensor' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:338:4: 'mjtStage' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:345:4: 'mjtDataType' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:121:4: 'mjtJoint' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:127:4: 'mjtGeom' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:149:4: 'mjtCamLight' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:157:4: 'mjtTexture' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:163:4: 'mjtIntegrator' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:168:4: 'mjtCollision' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:174:4: 'mjtCone' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:179:4: 'mjtJacobian' redeclared
[... 370 lines cut ...]
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:342:8: 'mjSTAGE_ACC' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:346:8: 'mjDATATYPE_REAL' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:347:8: 'mjDATATYPE_POSITIVE' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:348:8: 'mjDATATYPE_AXIS' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:349:8: 'mjDATATYPE_QUAT' redeclared
  running build_ext
  building 'mujoco_py.cymj' extension
  creating /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-87ikc8y0/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.62_37_macextensionbuilder
  creating /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-87ikc8y0/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.62_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7
[... cut ...] /generated/_pyxbld_1.50.1.62_37_macextensionbuilder/temp.macosx-10.9-x86_64-3.7/private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-87ikc8y0/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  /private/var/folders/p9/4tsd_xb54pq4fdbvckmx7ry80000gn/T/pip-install-87ikc8y0/mujoco-py/mujoco_py/cymj.c:658:10: fatal error: mjmodel.h: No such file or directory
   #include "mjmodel.h"
            ^~~~~~~~~~~
  compilation terminated.
  error: command '/usr/local/bin/gcc-8' failed with exit status 1

machinaut, 2018-10-19:
@marctheshark3 the builder expects the headers to be in you `$HOME/.mujoco/mjpro150/include` folder -- can you check that those files are there?

machinaut, 2018-10-22:
@marctheshark3 I can't tell from the image if that folder is in `$HOME/.mujoco/` but I'll trust you for that.

Do you get a new error with the latest version of mujoco-py?  I pushed an update last week which should print out if it fails to find the mujoco library in the expected location,

Fredeli, 2019-03-03:
Im having the same error. Any updates on this topic?

I double checked the paths and whether the headers exists in the `$HOME/.mujoco/mjpro150/include` directory.

ogail, 2019-03-25:
I had same problem and turned out that the dirs were correctly placed but the dir names were not as expected. The MuJoCo `2.0` version folder name starts with `mujoco` (which is what is added in the README file) however for `1.5` version folder name is expected to start with `mjpro` instead.

houyanxu, 2019-04-16:
Hi, I had the same problem. I  deal with it by following way.
Check the paths ~/.mujoco/mujoco200 whether have an _include_ directory. If not, copy the _include_ whole folder to the correct directory.

andrhua, 2019-05-19:
On Linux you probably wanna `7z x...`, not `7z e...`

adityagupte95, 2019-06-13:
[image]
I am facing the same issue where the mjmodel.h file cannot be located. But the file does exist at the desired location
[image]
I am using ubuntu 16.04, python 3.5.2.

mahon94, 2019-09-05:
@adityagupte95 for me moving everything inside "~/.mujoco/mujoco200/mujoco200_linux" to "~/.mujoco/mujoco200/" and deleting the mujoco200_linux fixed the problem

episodeyang, 2019-09-21:
```bash
export LD_LIBRARY_PATH=$HOME/.mujoco/mujoco200/bin:$LD_LIBRARY_PATH
export MUJOCO_PY_MJPRO_PATH=$HOME/.mujoco/mujoco200/
export MUJOCO_PY_MJKEY_PATH=$HOME/.mujoco/mjkey.txt
```

worked. In addition I had to do
```bash
brew uninstall gcc
brew uninstall gcc@8
brew uninstall gcc@9

brew install gcc@6
```

Only the gcc@6 worked with this installation. OSX version is 10.14.6 (18G87)

Kenivia, 2019-10-14:
> @adityagupte95 for me moving everything inside ~/.mujoco/mujoco200/mujoco200_linux to ~/.mujoco/mujoco200/ and deleting the mujoco200_linux fixed the problem

Same! the tutorial is kinda confusing

ZQX323, 2020-05-07:
> @adityagupte95 for me moving everything inside ~/.mujoco/mujoco200/mujoco200_linux to ~/.mujoco/mujoco200/ and deleting the mujoco200_linux fixed the problem

Great! Thanks a lot!

h3lio5, 2020-05-17:
> @adityagupte95 for me moving everything inside ~/.mujoco/mujoco200/mujoco200_linux to ~/.mujoco/mujoco200/ and deleting the mujoco200_linux fixed the problem

Thanks a lot! Your solution worked for me.

ffelten, 2022-07-12:
Hello, 

I got the same kind of error when trying to install on a custom directory in a container (Ubuntu 22, Python 3.9):

## Env
```
$ echo $MUJOCO_PY_MUJOCO_PATH/
/etc/bin/
LD_LIBRARY_PATH=...:/etc/bin/.mujoco/mujoco210/bin:...
```

## Install
```
$ ls /etc/bin/.mujoco/mujoco210/
THIRD_PARTY_NOTICES  bin  include  model  sample
```
## Test (first error)
```
$ pip install mujoco-py
... Installed in mujoco-py in /root/.local/lib/python3.9/site-packages
$ python
>>> import mujoco_py
```

`Missing path to your environment variable.
Current values LD_LIBRARY_PATH=...:/etc/bin/.mujoco/mujoco210/bin:...
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/etc/bin/bin` <- /bin/bin ???

## Test (second error)
```
$ LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/etc/bin/bin
$ python
>>> import mujoco_py
```

`/root/.local/lib/python3.9/site-packages/mujoco_py/cymj.c:744:10: fatal error: mjmodel.h: No such file or directory
  744 | #include "mjmodel.h"`

I have been searching for quite a while now and I think the modification of LD_LIBRARY_PATH is weird in the first place. Then, regarding the second error message. I have no clue on how to fix that,  I guess I should have a look at the includes?

Thanks in advance,

ffelten, 2022-07-13:
Okay, after a few more trial and errors:
It seems that setting `MUJOCO_PY_MUJOCO_PATH=/etc/bin/.mujoco/mujoco210` solves the problem.

clockzhong, 2023-10-14:
This is a bug reported in 2018, but why I encounter it also in 2023?

Deepakgthomas, 2024-01-09:
Any idea why this doesn't work - 
```
FROM ubuntu:22.04

WORKDIR /app

SHELL ["/bin/bash", "-c"] 

RUN echo "Hello World!"
RUN apt-get update && apt-get install -y \
    libosmesa6-dev \
    sudo \
    wget \
    curl \
    unzip \
    gcc \
    g++ \
    &&  apt-get install \
    libosmesa6-dev \
    && rm -rf /var/lib/apt/lists/*
    
ENV DEBIAN_FRONTEND=noninteractive

ENV PATH="/miniconda3/bin:${PATH}"
ARG PATH="/miniconda3/bin:${PATH}"
RUN cd / \
    && mkdir -p /miniconda3 \
    && wget https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh -O /miniconda3/miniconda.sh \
    && bash /miniconda3/miniconda.sh -b -u -p /miniconda3 \
    && /miniconda3/bin/conda init bash \
    && source ~/.bashrc \
    && conda init \
    && conda create -y -n myenv python=3.8 \
    && conda update -y conda
    

RUN wget https://roboti.us/download/mjpro150_linux.zip \
    && unzip mjpro150_linux.zip \
    && mkdir /mujoco \
    && mv mjpro150 /mujoco \
    && wget https://roboti.us/file/mjkey.txt \
    && mv mjkey.txt /mujoco \
    && rm mjpro150_linux.zip
    
RUN cd /miniconda3/envs/myenv/lib/ && mv libstdc++.so.6 libstdc++.so.6.old && ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 libstdc++.so.6

SHELL ["conda", "run", "-n", "myenv", "/bin/bash", "-c"]
EXPOSE 5003

RUN pip install --no-cache-dir "cython<3"
RUN pip install mujoco-py==1.50.1.0
```

michaelala25, 2024-06-11:
For anyone still encountering this, my issue was two-fold: I had accidentally downloaded aarch64 MuJoCo for an x86_64 system, and I was trying to use a MuJoCo version >2.10. Using the release provided in the README ([here](https://mujoco.org/download/mujoco210-linux-x86_64.tar.gz)) fixed this particular issue.