openai/mujoco-py #312: import error During handling of the above exception, another exception occurred, LockFailed

ZhanPython, 2018-10-20:
When I import mujoco_py, I got:

---------------------------------------------------------------------------
PermissionError                           Traceback (most recent call last)
~/anaconda3/lib/python3.6/site-packages/lockfile/linklockfile.py in acquire(self, timeout)
     18         try:
---> 19             open(self.unique_name, "wb").close()
     20         except IOError:

PermissionError: [Errno 13] Permission denied: '/home/zhanhx/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.63-py3.6.egg/mujoco_py/generated/zhanhx-Precision-M4600-3e312740.4595451864139058438194'

During handling of the above exception, another exception occurred:

LockFailed                                Traceback (most recent call last)
<ipython-input-2-85c6ae7fd0aa> in <module>()
----> 1 import mujoco_py

~/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.63-py3.6.egg/mujoco_py/__init__.py in <module>()
      1 #!/usr/bin/env python
      2 import os
----> 3 from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
      4 from mujoco_py.generated import const
      5 from mujoco_py.mjrenderpool import MjRenderPool

~/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.63-py3.6.egg/mujoco_py/builder.py in <module>()
    498 
    499 mjpro_path, key_path = discover_mujoco()
--> 500 cymj = load_cython_ext(mjpro_path)
    501 
    502 

~/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.63-py3.6.egg/mujoco_py/builder.py in load_cython_ext(mjpro_path)
     87     lockpath = os.path.join(os.path.dirname(cext_so_path), 'mujocopy-buildlock')
     88 
---> 89     with LockFile(lockpath):
     90         mod = None
     91         force_rebuild = os.environ.get('MUJOCO_PY_FORCE_REBUILD')

~/anaconda3/lib/python3.6/site-packages/lockfile/__init__.py in __enter__(self)
    195         Context manager support.
    196         """
--> 197         self.acquire()
    198         return self
    199 

~/anaconda3/lib/python3.6/site-packages/lockfile/linklockfile.py in acquire(self, timeout)
     19             open(self.unique_name, "wb").close()
     20         except IOError:
---> 21             raise LockFailed("failed to create %s" % self.unique_name)
     22 
     23         timeout = timeout if timeout is not None else self.timeout

LockFailed: failed to create /home/zhanhx/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.63-py3.6.egg/mujoco_py/generated/zhanhx-Precision-M4600-3e312740.4595451864139058438194

How to tackle this? Many thanks.

ZhanPython, 2018-10-20:
when I do pip uninstall mujoco-py

I got:
Exception:
Traceback (most recent call last):
  File "/home/zhanhx/anaconda3/lib/python3.6/shutil.py", line 544, in move
    os.rename(src, real_dst)
PermissionError: [Errno 13] Permission denied: '/home/zhanhx/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.63-py3.6.egg' -> '/tmp/pip-uninstall-haqc94kp/home/zhanhx/anaconda3/lib/python3.6/site-packages/mujoco_py-1.50.1.63-py3.6.egg'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zhanhx/anaconda3/lib/python3.6/site-packages/pip/_internal/cli/base_command.py", line 143, in main
    status = self.run(options, args)
  File "/home/zhanhx/anaconda3/lib/python3.6/site-packages/pip/_internal/commands/uninstall.py", line 75, in run
    auto_confirm=options.yes, verbose=self.verbosity > 0,
  File "/home/zhanhx/anaconda3/lib/python3.6/site-packages/pip/_internal/req/req_install.py", line 683, in uninstall
    uninstalled_pathset.remove(auto_confirm, verbose)
  File "/home/zhanhx/anaconda3/lib/python3.6/site-packages/pip/_internal/req/req_uninstall.py", line 224, in remove
    renames(path, new_path)
  File "/home/zhanhx/anaconda3/lib/python3.6/site-packages/pip/_internal/utils/misc.py", line 280, in renames
    shutil.move(old, new)
  File "/home/zhanhx/anaconda3/lib/python3.6/shutil.py", line 556, in move
    rmtree(src)
  File "/home/zhanhx/anaconda3/lib/python3.6/shutil.py", line 480, in rmtree
    _rmtree_safe_fd(fd, path, onerror)
  File "/home/zhanhx/anaconda3/lib/python3.6/shutil.py", line 418, in _rmtree_safe_fd
    _rmtree_safe_fd(dirfd, fullname, onerror)
  File "/home/zhanhx/anaconda3/lib/python3.6/shutil.py", line 438, in _rmtree_safe_fd
    onerror(os.unlink, fullname, sys.exc_info())
  File "/home/zhanhx/anaconda3/lib/python3.6/shutil.py", line 436, in _rmtree_safe_fd
    os.unlink(name, dir_fd=topfd)
PermissionError: [Errno 13] Permission denied: 'dependency_links.txt'

machinaut, 2018-10-22:
It looks like you don't have permission to edit files in your python installation.

This looks like it could be the cause of both the import failure, and the uninstall failure.

If you make sure all of the relevant files have correct permissions and try again, what happens?

closed 2018-11-24 by ZhanPython

jamesheald, 2019-12-25:
I received the following error when trying to import mujoco_py:

PermissionError: [Errno 13] Permission denied: b'/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/mujoco_py/generated/mujocopy-buildlock'

I solved as follows:

cd /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/mujoco_py
sudo chmod -R 777 ./ (change permission of all files in all subfolders recursively)