openai/mujoco-py #313: Adding deformable objects to model

harjatinsingh, 2018-10-20:
I trying to add a deformable object(eg. cloth) to mujoco model, using the documentation under 'composite ' element in mujoco modelling (http://mujoco.org/book/modeling.html). Following is my xml right now 

```
<?xml version="1.0" ?>
<mujoco>
	 <compiler inertiafromgeom='auto' coordinate="global" />

	<worldbody>
     	<light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>  
      	<camera name="camera1" pos="0 0 0.25" />
        	<body>
        	<composite type="grid" count="9 9 1" spacing="0.05" offset="0 0 1">
        		<skin material="matcarpet" inflate="0.001" subgrid="3" texcoord="true"/>
        		<geom size=".02"/>
        		<pin coord="0 0"/>
        		<pin coord="8 0"/>
    		</composite>
    		<body>
   </worldbody>

</mujoco> 
```

On doing `load_model_from_path(xml_path)` I get the following error :

> 
`mj_loadXML error: b"XML Error: Schema violation: unrecognized element\n\nElement 'composite', line 9, column 13\n"`

Does mujoco-py support deformable objects yet ?

machinaut, 2018-10-22:
Mujoco-py supports mujoco 1.50, which does not support deformable objects https://www.roboti.us/index.html#mjpro150

machinaut, 2018-10-22:
Relevant issue to follow is here!  https://github.com/openai/mujoco-py/issues/303

closed 2018-10-22 by machinaut