openai/mujoco-py #314: ModuleNotFoundError: No module named 'lockfile' (mujoco-py-1.50.1.65)

tugceakin, 2018-10-24:
One of my builds use `gym[all]==0.9.6` which depends on mujoco-py and it
has been failing recently. I can get it working by specifically using `mujoco-py-1.50.1.59`, but not sure why I'm getting below errors when installing mujoco-py-1.50.1.65.

Step #2 - "gpu":   Running setup.py bdist_wheel for mujoco-py: started
Step #2 - "gpu":   Running setup.py bdist_wheel for mujoco-py: finished with status 'error'
Step #2 - "gpu":   Complete output from command /opt/conda/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-a41hjscy/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmptd9681xipip-wheel- --python-tag cp36:
Step #2 - "gpu":   running bdist_wheel
Step #2 - "gpu":   running build
Step #2 - "gpu":   Traceback (most recent call last):
Step #2 - "gpu":     File "<string>", line 1, in <module>
Step #2 - "gpu":     File "/tmp/pip-build-a41hjscy/mujoco-py/setup.py", line 44, in <module>
Step #2 - "gpu":       tests_require=read_requirements_file('requirements.dev.txt'),
Step #2 - "gpu":     File "/opt/conda/lib/python3.6/site-packages/setuptools/__init__.py", line 129, in setup
Step #2 - "gpu":       return distutils.core.setup(**attrs)
Step #2 - "gpu":     File "/opt/conda/lib/python3.6/distutils/core.py", line 148, in setup
Step #2 - "gpu":       dist.run_commands()
Step #2 - "gpu":     File "/opt/conda/lib/python3.6/distutils/dist.py", line 955, in run_commands
Step #2 - "gpu":       self.run_command(cmd)
Step #2 - "gpu":     File "/opt/conda/lib/python3.6/distutils/dist.py", line 974, in run_command
Step #2 - "gpu":       cmd_obj.run()
Step #2 - "gpu":     File "/opt/conda/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 204, in run
Step #2 - "gpu":       self.run_command('build')
Step #2 - "gpu":     File "/opt/conda/lib/python3.6/distutils/cmd.py", line 313, in run_command
Step #2 - "gpu":       self.distribution.run_command(command)
Step #2 - "gpu":     File "/opt/conda/lib/python3.6/distutils/dist.py", line 974, in run_command
Step #2 - "gpu":       cmd_obj.run()
Step #2 - "gpu":     File "/tmp/pip-build-a41hjscy/mujoco-py/setup.py", line 28, in run
Step #2 - "gpu":       import mujoco_py  # noqa: force build
Step #2 - "gpu":     File "/tmp/pip-build-a41hjscy/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
Step #2 - "gpu":       from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
Step #2 - "gpu":     File "/tmp/pip-build-a41hjscy/mujoco-py/mujoco_py/builder.py", line 20, in <module>
Step #2 - "gpu":       from lockfile import LockFile
Step #2 - "gpu":   ModuleNotFoundError: No module named 'lockfile'
Step #2 - "gpu":   
Step #2 - "gpu":   ----------------------------------------
Step #2 - "gpu": [91m  Failed building wheel for mujoco-py
Step #2 - "gpu": [0m  Running setup.py clean for mujoco-py
Step #2 - "gpu":   Running setup.py bdist_wheel for glfw: started
Step #2 - "gpu":   Running setup.py bdist_wheel for glfw: finished with status 'done'
Step #2 - "gpu":   Stored in directory: /root/.cache/pip/wheels/ca/aa/05/9b802212ed86ef800f54025059bac9e64f58dfaf17fb97e94e
Step #2 - "gpu": Successfully built dronekit gym box2d jieba pinyin pycurl pymavlink python-speech-features snownlp future PyOpenGL Box2D-kengz glfw
Step #2 - "gpu": Failed to build mujoco-py

machinaut, 2018-10-25:
The python package `lockfile` is required -- and should be required https://github.com/openai/mujoco-py/blob/master/requirements.txt#L6

machinaut, 2018-10-25:
`pip install lockfile` should be enough to fix your specific issue.

Not sure why installing gym isn't picking this up, though.

machinaut, 2018-10-25:
@tugceakin if this is an open source project could you link to the build scripts (or share an example)?  It'd be useful to be able to reproduce this.

machinaut, 2018-10-26:
Made a small reproduction dockerfile, which at least works for me, and might be useful as a reference https://gist.github.com/machinaut/b5a7c6719a30e9fbfd95c13d5e963b61