openai/mujoco-py #316: Failed to load xml file

GbengaOdesanmi, 2018-10-25:
Traceback (most recent call last):
  File "disco_fetch.py", line 9, in <module>
    model = load_model_from_path("xmls/fetch/main.xml")
  File "cymj.pyx", line 131, in mujoco_py.cymj.load_model_from_path
Exception: Failed to load XML file: xmls/fetch/main.xml. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'

_Originally posted by @GbengaOdesanmi in https://github.com/openai/mujoco-py/issue_comments#issuecomment-433088647_

machinaut, 2018-10-25:
The examples are meant to be run from the root directory.

If you are in the root directory and run `python examples/disco_fetch.py` does it work?

GbengaOdesanmi, 2018-10-29:
Thanks it was path problem @machinaut

closed 2018-10-29 by GbengaOdesanmi