openai/mujoco-py #318: It gives segmentation fault and error messages related to glfw when I render Reacher-v2

haanvid, 2018-10-29:
It gives segmentation fault and error messages related to glfw when I render Reacher-v2

The error message:

`Creating window glfw
[haanvid-server:02719] *** Process received signal ***
[haanvid-server:02719] Signal: Segmentation fault (11)
[haanvid-server:02719] Signal code: Address not mapped (1)
[haanvid-server:02719] Failing at address: (nil)
[haanvid-server:02719] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x12890)[0x7f7ab5585890]
[haanvid-server:02719] *** End of error message ***
Segmentation fault (core dumped)
`

My conda environment has:
`# Name                    Version                   Build  Channel
_tflow_1100_select        0.0.1                       gpu  
_tflow_180_select         1.0                         gpu  
absl-py                   0.4.1                    py35_0  
astor                     0.7.1                    py35_0  
atari-py                  0.1.6                     <pip>
baselines                 0.1.5                     <pip>
blas                      1.0                         mkl  
bleach                    1.5.0                    py35_0  
bzip2                     1.0.6                h470a237_2    conda-forge
ca-certificates           2018.03.07                    0  
cairo                     1.14.6                        5    conda-forge
certifi                   2018.8.24                py35_1  
cffi                      1.11.5                    <pip>
chardet                   3.0.4                     <pip>
Click                     7.0                       <pip>
cloudpickle               0.6.1                     <pip>
cudatoolkit               8.0                           3  
cudnn                     7.1.3                 cuda8.0_0  
cupti                     8.0.61                        0  
Cython                    0.29                      <pip>
dbus                      1.13.0               h3a4f0e9_0    conda-forge
dill                      0.2.8.2                   <pip>
expat                     2.2.5                hfc679d8_2    conda-forge
ffmpeg                    3.2.4                hf82bc7d_4    conda-forge
fontconfig                2.12.1                        6    conda-forge
freetype                  2.7                           2    conda-forge
future                    0.16.0                   py35_2  
future                    0.16.0                    <pip>
gast                      0.2.0                    py35_0  
gettext                   0.19.8.1             h5e8e0c9_1    conda-forge
giflib                    5.1.4                h470a237_1    conda-forge
glfw                      1.7.0                     <pip>
glfw3                     3.2.1                         0    menpo
glib                      2.51.4                        0    conda-forge
grpcio                    1.12.1           py35hdbcaa40_0  
gst-plugins-base          1.8.0                         0    conda-forge
gstreamer                 1.8.0                         2    conda-forge
gym                       0.10.8                    <pip>
harfbuzz                  1.3.4                         2    conda-forge
hdf5                      1.10.2               hba1933b_1  
html5lib                  0.9999999                py35_0  
icu                       58.2                 hfc679d8_0    conda-forge
idna                      2.7                       <pip>
imageio                   2.4.1                     <pip>
intel-openmp              2019.0                      118  
jasper                    1.900.1              hff1ad4c_5    conda-forge
joblib                    0.12.5                    <pip>
jpeg                      9c                   h470a237_1    conda-forge
libffi                    3.2.1                hfc679d8_5    conda-forge
libgcc                    7.2.0                h69d50b8_2  
libgcc-ng                 8.2.0                hdf63c60_1  
libgcrypt                 1.7.3                         0    clinicalgraphics
libgcrypt11               1                             1    clinicalgraphics
libgfortran               3.0.0                         1    conda-forge
[... 98 lines cut ...]
mpi4py                 3.0.0     
mujoco-py              1.50.1.65 /home/haanvid/mujoco-py
numpy                  1.14.5    
opencv-python          3.4.3.18  
Pillow                 5.3.0     
pip                    10.0.1    
progressbar2           3.38.0    
protobuf               3.6.0     
pycparser              2.19      
pyglet                 1.3.2     
PyOpenGL               3.1.0     
python-utils           2.3.0     
requests               2.20.0    
scipy                  1.1.0     
setuptools             39.1.0    
six                    1.11.0    
tensorboard            1.10.0    
tensorflow             1.10.0    
tensorflow-tensorboard 1.5.1     
termcolor              1.1.0     
tqdm                   4.28.1    
urllib3                1.24      
Werkzeug               0.14.1    
wheel                  0.31.1    
`

machinaut, 2018-10-29:
Is the render call sent by the main thread or a different thread?

In the past we've seen some renderer crashes based on what thread they were running on.

haanvid, 2018-10-29:
It turns out the missing libraries causing the errors are already installed in my virtual env.   I've preloaded those using my bashrc as suggested by the ppl who suffered from the same problem (glew, glfw)... It works fine now.

closed 2018-10-29 by haanvid