openai/mujoco-py #319: Camera Calibration

xinyiren, 2018-10-31:
Hi,

I wonder if mujoco-py supports passing camera intrinsic matrix into the mujoco camera?

I have a logitech C920 camera. I also have the intrinsic matrix, distortion params, rectification matrix and extrinsic matrix generated using ROS camera calibration. Is there any way that I can manipulate/pass these camera calibration data into the mujoco camera?

machinaut, 2018-11-01:
I think this is a good question!  The MuJoCo forums are probably the right place to ask this http://www.mujoco.org/forum/index.php

(This package is just the python wrapper for the mujoco library)

Looking at the documentation for the mujoco camera renderer here http://www.mujoco.org/book/APIreference.html#mjvCamera  I don't see those settings -- but it might be worth looking through the mujoco documentation to see if it's mentioned elsewhere.

I've only ever used mujoco cameras as idealized/pinholes.

machinaut, 2018-11-01:
Closing this for now as the question is about the underlying library.

closed 2018-11-01 by machinaut