openai/mujoco-py #322: libGL error: No matching fbConfigs or visuals found

Tina-ting, 2018-11-09:
**Describe the bug**
Hi, when I try to test the installation of Mujoco, I got the following errors:
MuJoCo Pro version 2.00
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
ERROR: could not initialize GLFW
(no other errors )

**To Reproduce**
My steps:
1) mkdir ~/.mujoco200
2) cp mujoco200_linux.zip ~/.mujoco200
3) cd ~/.mujoco200
   unzip mujoco200_linux.zip
4) cp mjkey.txt ~/.mujoco200
   cp mjkey.txt ~/.mujoco200/mujoco200_linux/bin
5) export LD_LIBRARY_PATH=~/.mujoco200/mujoco200_linux/bin${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
export MUJOCO_KEY_PATH=~/.mujoco200${MUJOCO_KEY_PATH}
6) cd ~/.mujoco200/mujoco200_linux/bin
7) ./simulate ../model/humanoid.xml

**Expected behavior**
There should be a window to appear, but nothing appeared. 

**Desktop (please complete the following information):**
 - Mujoco Version:200
- Ubuntu 16.04.5 LTS (GNU/Linux 4.15.0-36-generic x86_64)
- CUDA Version 9.0.176
- Python 3.6.6 |Anaconda, Inc.| (default
- conda list: glfw 1.7.0,  PyOpenGL 3.1.0

**Additional context**
I have tried mujoco150 but with same errors. How to tackle this? Many thanks!

machinaut, 2018-11-09:
@Tina-2018 mujoco200 does not work with this package, only mujoco150 for now.

Those errors seem like an issue with OpenGL (graphics) drivers.  Do you know what graphics drivers you are using (nvidia, mesa, etc)?

Specifically testing the GLFW library, can you try just importing it from python `python -c "import glfw"` or try running some simple PyGLFW code to see if that works?

Also, since this is an issue with the underlying mujoco library, another place you can ask questions is in the mujoco forum.

This question and answer in particular seems related http://www.mujoco.org/forum/index.php?threads/error-glew-initalization-error-missing-gl-version.3717/#post-4768

machinaut, 2018-11-09:
@Tina-2018 I just tested this dockerfile https://gist.github.com/machinaut/b5a7c6719a30e9fbfd95c13d5e963b61 with ubuntu 16.04 and it worked.

Can you check if you have those ubuntu packages installed and try again after installing them?

Tina-2018, 2018-11-11:
@machinaut  My driver=nvidia, and "import glfw" in python returns no error.
Because I am one of the users of an Ubuntu client with limited user rights, and I am not familiar with docker container, So I just try step by step as guided by your tutorial. Seems Mujoco150 did not work too. 
I guess the problem may be caused by the OpenGL driver, and I will try to find solutions in mujoco forum.
Thanks so much for your help!

yunzhe-tao, 2020-03-11:
@TinaZhangTing Have you figured out the solution? I have exactly the same issue