openai/mujoco-py #323: Installation error: "libgcrypt.so.11: cannot open shared object file"

kyr-pol, 2018-11-09:
Hi everyone, trying to install mujoco-py on a server running Fedora 27 using Anaconda. I couldn't find anything online for a similar error and I run out of ideas so here is the traceback:
```
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-r9ds_18e/mujoco-py/setup.py", line 44, in <module>
      tests_require=read_requirements_file('requirements.dev.txt'),
    File "/usr/lib/python3.6/site-packages/setuptools/__init__.py", line 129, in setup
      return distutils.core.setup(**attrs)
    File "/data/greyostrich/aims/aims15/polymena/anaconda3/envs/tf_gpu36/lib/python3.6/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/data/greyostrich/aims/aims15/polymena/anaconda3/envs/tf_gpu36/lib/python3.6/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/data/greyostrich/aims/aims15/polymena/anaconda3/envs/tf_gpu36/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/usr/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 202, in run
      self.run_command('build')
    File "/data/greyostrich/aims/aims15/polymena/anaconda3/envs/tf_gpu36/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/data/greyostrich/aims/aims15/polymena/anaconda3/envs/tf_gpu36/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-r9ds_18e/mujoco-py/setup.py", line 28, in run
      import mujoco_py  # noqa: force build
    File "/tmp/pip-install-r9ds_18e/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
      from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
    File "/tmp/pip-install-r9ds_18e/mujoco-py/mujoco_py/builder.py", line 503, in <module>
      cymj = load_cython_ext(mjpro_path)
    File "/tmp/pip-install-r9ds_18e/mujoco-py/mujoco_py/builder.py", line 106, in load_cython_ext
      mod = load_dynamic_ext('cymj', cext_so_path)
    File "/tmp/pip-install-r9ds_18e/mujoco-py/mujoco_py/builder.py", line 124, in load_dynamic_ext
      return loader.load_module()
  ImportError: libgcrypt.so.11: cannot open shared object file: No such file or directory
```

Any ideas? Anyone with similar issues?

machinaut, 2018-11-09:
I haven't tried fedora in a while, but my guess would be missing prerequisite packages.  Have you tried installing the package that provides `libgcrypt.so.11` and trying again?

machinaut, 2018-11-09:
A year ago I had mujoco-py working in fedora and these were the packages I needed https://gist.github.com/machinaut/f5cf076b33752334a5618ed906baf23a

If you install those packages does it work for you?

kyr-pol, 2018-11-10:
Hey @machinaut thanks for the advice, I'll give this a shot. It might be a bit harder in my case since this a server I don't have root privileges to. For some of the packages I found conda versions, for some not, I might have to try to install from source.

HareshMiriyala, 2020-04-11:
the clinicalgraphics channel is no longer available. I still get this error "ImportError: libgcrypt.so.11: cannot open shared object file: No such file or directory". 

Installing mujoco_py on a compute cluster without sudo access, using conda.

kyr-pol, 2020-04-23:
Hey @HareshMiriyala ,

I stumbled upon the same problem myself again, and didn't manage to find any conda workarounds this time but installing with rpm without root access was a lot more painless than I thought. Basically I followed the suggestions [here](https://github.com/openai/mujoco-py/issues/96#issuecomment-493391273) but for libcrypt11.

First you download (and save) the libcrypt11 tar ball with something like:
`curl -o libgcrypt11.rpm ftp://ftp.pbone.net/mirror/ftp5.gwdg.de/pub/opensuse/repositories/home:/bosconovic:/branches:/home:/elimat:/lsi/openSUSE_Leap_15.1/x86_64/libgcrypt11-1.5.4-lp151.23.29.x86_64.rpm`

Then install locally with this:

`rpm2cpio libgcrypt11.rpm | cpio -id`

which creates a usr folder in your current path, and then all you have to do is add the folders to the environment variables:
` export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/YOUR_PATH/usr/lib64`
`export LDFLAGS="-L/YOUR_PATH/usr/lib64"`

`pip install mujoco_py` worked fine for me after that.

GoingMyWay, 2021-11-13:
> Hey @HareshMiriyala ,
>
> I stumbled upon the same problem myself again, and didn't manage to find any conda workarounds this time but installing with rpm without root access was a lot more painless [...]

I tried to install, but it returned this error

```
$ rpm2cpio libgcrypt11.rpm | cpio -id
rpm2cpio: symbol lookup error: /lib64/librpmio.so.8: undefined symbol: EVP_md2, version OPENSSL_1_1_0
cpio: premature end of archive
```

GoingMyWay, 2021-11-13:
## Build from source

Step 1. Download the source

Go to: https://www.gnupg.org/ftp/gcrypt/libgcrypt/

and download https://www.gnupg.org/ftp/gcrypt/libgcrypt/libgcrypt-1.5.3.tar.gz

Step 2. Create a local dir to build and install by following the instructions in this link:https://www.linuxfromscratch.org/blfs/view/svn/general/libgcrypt.html

Step 3. Put the include and lib path to your environment variables

caplett, 2022-09-10:
> ## Build from source
> 
> Step 1. Download the source
> 
> Go to: https://www.gnupg.org/ftp/gcrypt/libgcrypt/
> 
> and download https://www.gnupg.org/ftp/gcrypt/libgcrypt/libgcrypt-1.5.3.tar.gz
> 
> Step 2. Create a local dir to build and install by following the instructions in this link:https://www.linuxfromscratch.org/blfs/view/svn/general/libgcrypt.html
> 
> untar and go the untarred directory
> 
> ```
> ./configure --prefix=/home/USER_NAME/opt/lib/libgcrypt && make
> ```
> 
> and
> 
> ```
> make install &&
> ```
> 
> Step 3. Put the include and lib path to your environment variables
> 
> ```
> export PATH="/home/USER_NAME/opt/lib/libgcrypt/bin:$PATH"
> export C_INCLUDE_PATH="/home/USER_NAME/opt/lib/libgcrypt/include:$C_LIBRARY_PATH:$C_INCLUDE_PATH"
> ```

I had still problems where compilation did not work.
For me this did the trick as a "Step 4":
```
export LD_PRELOAD=$/home/USER_NAME/opt/lib/libgcrypt/lib/libgcrypt.so.11 
```

I dont know if this has untended side-effects