openai/mujoco-py #326: error when import mujoco_py on 64bits windows10

Liliduan, 2018-11-19:
64bits windows10
mjpro150
mujoco_py:1.50.1
anaconda3-5.2.0
After installing the mujoco_py on  windows, I tried to import mujoco_py on cmd ,but failed. 
The details about the error I met:
>>> import mujoco_py
Import error. Trying to rebuild mujoco_py.
running build_ext
building 'mujoco_py.cymj' extension
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -ID:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py -IC:\Users\lld_4\.mujoco\mjpro150\include -ID:\software\anaconda\lib\site-packages\numpy\core\include -ID:\software\anaconda\include -ID:\software\anaconda\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-ID:\Windows Kits\10\include\10.0.17763.0\ucrt" "-ID:\Windows Kits\10\include\10.0.17763.0\shared" "-ID:\Windows Kits\10\include\10.0.17763.0\um" "-ID:\Windows Kits\10\include\10.0.17763.0\winrt" "-ID:\Windows Kits\10\include\10.0.17763.0\cppwinrt" /TcD:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\cymj.c /FoD:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.68_36_windowsextensionbuilder\temp.win-amd64-3.6\Release\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\cymj.obj -fopenmp -w
cl: 命令行 warning D9025 :正在重写“/W3”(用“/w”)
cl: 命令行 warning D9002 :忽略未知选项“-fopenmp”
cymj.c
d:\software\anaconda\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(12) : Warning Msg: Using deprecated NumPy API, disable it by #defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -ID:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py -IC:\Users\lld_4\.mujoco\mjpro150\include -ID:\software\anaconda\lib\site-packages\numpy\core\include -ID:\software\anaconda\include -ID:\software\anaconda\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-ID:\Windows Kits\10\include\10.0.17763.0\ucrt" "-ID:\Windows Kits\10\include\10.0.17763.0\shared" "-ID:\Windows Kits\10\include\10.0.17763.0\um" "-ID:\Windows Kits\10\include\10.0.17763.0\winrt" "-ID:\Windows Kits\10\include\10.0.17763.0\cppwinrt" /TcD:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py/gl/dummyshim.c /FoD:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.68_36_windowsextensionbuilder\temp.win-amd64-3.6\Release\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py/gl/dummyshim.obj -fopenmp -w
cl: 命令行 warning D9025 :正在重写“/W3”(用“/w”)
cl: 命令行 warning D9002 :忽略未知选项“-fopenmp”
dummyshim.c
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:C:\Users\lld_4\.mujoco\mjpro150\bin /LIBPATH:D:\software\anaconda\libs /LIBPATH:D:\software\anaconda\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\lib\um\x64" "/LIBPATH:D:\Windows Kits\10\lib\10.0.17763.0\ucrt\x64" "/LIBPATH:D:\Windows Kits\10\lib\10.0.17763.0\um\x64" mujoco150.lib /EXPORT:PyInit_cymj D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.68_36_windowsextensionbuilder\temp.win-amd64-3.6\Release\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\cymj.obj D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.68_36_windowsextensionbuilder\temp.win-amd64-3.6\Release\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py/gl/dummyshim.obj /OUT:D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.68_36_windowsextensionbuilder\lib.win-amd64-3.6\mujoco_py\cymj.cp36-win_amd64.pyd /IMPLIB:D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.68_36_windowsextensionbuilder\temp.win-amd64-3.6\Release\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\cymj.cp36-win_amd64.lib -fopenmp
LINK : warning LNK4044: 无法识别的选项“/fopenmp”；已忽略
LINK : fatal error LNK1104: 无法打开文件“D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\generated\_pyxbld_1.50.1.68_36_windowsextensionbuilder\temp.win-amd64-3.6\Release\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\cymj.cp36-win_amd64.lib”
Traceback (most recent call last):
  File "D:\software\anaconda\lib\distutils\_msvccompiler.py", line 519, in link
    self.spawn([self.linker] + ld_args)
  File "D:\software\anaconda\lib\distutils\_msvccompiler.py", line 542, in spawn
    return super().spawn(cmd)
  File "D:\software\anaconda\lib\distutils\ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "D:\software\anaconda\lib\distutils\spawn.py", line 38, in spawn
    _spawn_nt(cmd, search_path, dry_run=dry_run)
  File "D:\software\anaconda\lib\distutils\spawn.py", line 81, in _spawn_nt
    "command %r failed with exit status %d" % (cmd, rc))
distutils.errors.DistutilsExecError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1104

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\builder.py", line 503, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\builder.py", line 243, in _build_impl
    dist.run_commands()
  File "D:\software\anaconda\lib\distutils\dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "D:\software\anaconda\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "D:\software\anaconda\lib\site-packages\Cython\Distutils\old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "D:\software\anaconda\lib\distutils\command\build_ext.py", line 339, in run
    self.build_extensions()
  File "D:\software\anaconda\lib\site-packages\mujoco_py-1.50.1.68-py3.6.egg\mujoco_py\builder.py", line 143, in build_extensions
    build_ext.build_extensions(self)
  File "D:\software\anaconda\lib\site-packages\Cython\Distutils\old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "D:\software\anaconda\lib\distutils\command\build_ext.py", line 558, in build_extension
    target_lang=language)
  File "D:\software\anaconda\lib\distutils\ccompiler.py", line 717, in link_shared_object
    extra_preargs, extra_postargs, build_temp, target_lang)
  File "D:\software\anaconda\lib\distutils\_msvccompiler.py", line 522, in link
    raise LinkError(msg)
distutils.errors.LinkError: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1104

Is there anyone knows how to cope with this issue?

machinaut, 2018-11-19:
I haven't seen this error before.  Are you able to link and load other cython programs on the same computer?

wp92, 2018-12-01:
I met this error, sincerely hope that someone can cope with the issue@

DarioUrbina, 2020-04-30:
same error... any suggestions?

duanwenbo, 2021-08-02:
same +1

zhshao17, 2024-08-06:
same error, i wanna install mujoco-py in windows10 with vs2017, anyone have suggestions?

AuroraHashcat, 2024-09-05:
same error on win11 with python3.7. already tried installing build tools 2015 (but just installation, I don't know if there should be more action), vs2022, set the environment variables, shorten the path name length. By the way there is another issue pointing out that -fopen should be corrected as /fopen for stablity in c file. I tried all but no use.