openai/mujoco-py #327: Unable to provide custom XML for mujoco-py

cyrilzakka, 2018-11-19:
Hello,

I hope you're doing well. I'm having an issue loading a custom XML model into `mujoco-py`, and by extension `gym`. I've copied `pick_and_place.xml` and `shared.xml` into a separate directory along with all of their assets and have modified the color of the background to test my setup. Doing the following with the `humanoid.xml` file works well.

```
import mujoco_py
import os
mj_path, _ = mujoco_py.utils.discover_mujoco()
xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
model = mujoco_py.load_model_from_path(xml_path)
```

Replacing it with my own XML file (using the full file path) as so:
`xml_path = os.path.join(mj_path, 'model', '/Users/cyril/gym-assist/gym_assist/envs/assets/handoff/pick_and_place.xml')`

yields the following error: 
```
Exception: Failed to load XML file: /Users/cyril/gym-assist/gym_assist/envs/assets/handoff/pick_and_place.xml. mj_loadXML error: b"XML Error: Include error: 'XML parse error at line 0, column 0:\nFailed to open file\n'\nElement 'include', line 16, column 9\n"
```

Any ideas?

closed 2018-11-19 by cyrilzakka

nohboogy, 2019-08-08:
Did you solve this problem? I'm also finding the solution also

BlueShit-1120, 2021-05-24:
> Did you solve this problem? I'm also finding the solution also

Have you ever solve this problem? I also met it. Could you please help me with this?

BlueShit-1120, 2021-05-24:
> Hello,
> 
> I hope you're doing well. I'm having an issue loading a custom XML model into `mujoco-py`, and by extension `gym`. I've copied `pick_and_place.xml` and `shared.xml` into a separate directory along with all of their assets and have modified the color of the background to test my setup. Doing the following with the `humanoid.xml` file works well.
> 
> ```
> import mujoco_py
> import os
> mj_path, _ = mujoco_py.utils.discover_mujoco()
> xml_path = os.path.join(mj_path, 'model', 'humanoid.xml')
> model = mujoco_py.load_model_from_path(xml_path)
> ```
> 
> Replacing it with my own XML file (using the full file path) as so:
> `xml_path = os.path.join(mj_path, 'model', '/Users/cyril/gym-assist/gym_assist/envs/assets/handoff/pick_and_place.xml')`
> 
> yields the following error:
> 
> ```
> Exception: Failed to load XML file: /Users/cyril/gym-assist/gym_assist/envs/assets/handoff/pick_and_place.xml. mj_loadXML error: b"XML Error: Include error: 'XML parse error at line 0, column 0:\nFailed to open file\n'\nElement 'include', line 16, column 9\n"
> ```
> 
> EDIT 1:
> Not modifying anything also leads to the error. Going to look deeper into it.
> 
> Any ideas?

Have you ever solve this problem? I also met it. Could you please help me with this? Thanks.

EunHooLee, 2022-10-31:
Did you solve this problem? I'm  also finding the solution also - 2

```
Traceback (most recent call last):
  File "mujoco_py_test_2.py", line 13, in <module>
    model = mujoco_py.load_model_from_path("/home/look-closer/src/env/robot/assets/robot/push.xml")
  File "mujoco_py/cymj.pyx", line 175, in mujoco_py.cymj.load_model_from_path
Exception: Failed to load XML file: /home/look-closer/src/env/robot/assets/robot/push.xml. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'
```