openai/mujoco-py #328: Custom mesh jittering in Mujoco environment in OpenAI gym

cyrilzakka, 2018-11-19:
No clue if this belongs here but... I've tried modifying the `FetchPickAndPlace-v1` OpenAI environment to replace the cube with a pair of scissors. Everything works perfectly except for the fact that my custom mesh seems to jitter a few millimeters in and out of the table every few time steps. I've included a picture mid-jitter below:

[image]

As you can see, the scissors are caught mid-way through the surface of the table. How can I prevent this? All I've done is switch out the code for the cube in `pick_and_place.xml` with the asset related to the scissor mesh. Here's the code of interest:
```
<body name="object0" pos="0.0 0.0 0.0">
            <joint name="object0:joint" type="free" damping="0.01"></joint>
            <geom size="0.025 0.025 0.025" mesh="tool0:scissors" condim="3" name="object0" material="tool_mat" class="tool0:matte" mass="2"></geom>
            <site name="object0" pos="0 0 0" size="0.02 0.02 0.02" rgba="1 0 0 1" type="sphere"></site>
</body>
```

I've tried playing around with the coordinates of the position and geometry but to no avail. Any tips? Replacing `mesh="tool0:scissors"` with `type="box"` gets rid of the problem entirely but I'm back to square one.

machinaut, 2018-11-19:
This seems to be more of an issue with physics simulation than python bindings, so I would recommend also asking it on the MuJoCo forums!

For this problem in particular, it's possible you want to increase the amount of energy that can be removed during contacts (soft/absorptive contacts) -- otherwise the bouncing might just be the initial potential energy from it sitting above the table that never goes away.

It's worth visualizing the convexification of the mesh as well, since this is a preprocessing step mujoco does in order to get a convex geom to make contacts with.

cyrilzakka, 2018-11-19:
Thank you for the quick reply - I need to learn about half of the things you just mentioned but appreciate the response nonetheless. I'll leave this open until I find a solution, which I'll post here then close.

Cheers!

cyrilzakka, 2018-11-19:
As suggested by Emo Todorov in the [MuJoCo forums](http://www.mujoco.org/forum/index.php?threads/custom-mesh-jittering-in-mujoco-environment-in-openai-gym.3860/#post-5074):

 1. Replace the ground box with a plane (or is it already a plane?) and use MuJoCo 2.0. The latest version of the collision detector generates multiple contacts between a mesh and a plane, which results in more stable simulation. But this only works for plane-mesh, not for box-mesh.

2. The better solution is to break the mesh into several meshes, and include them as multiple geoms in the same body. Then MuJoCo will construct the convex hull of each sub-mesh, resulting in multiple contact points (even without the special plane mechanism mentioned above) and furthermore it will be a better approximation to the actual object geometry.

closed 2018-11-20 by cyrilzakka