openai/mujoco-py #330: Graphics not showing on Mac OS X

xqphd, 2018-11-23:
**Description**
I have installed all the packages & dependencies following the document, including Mujoco150 with student license. When I run the example code CartPole on https://gym.openai.com/docs/#observations, it produces text outputs normally, no error message is given. 

But no graphic is shown! And Only a blank window is shown to me. 

[image]

So, what's the problem and how to solve it please?

**To Reproduce**
Running some example code, and then I see the blank window. 

**Expected behavior**
A graphical window which shows exactly what is given on openai.  

**Error Messages**
No error message is reported. 

**Desktop (please complete the following information):**
 - OS: macOS 10.11.1
 - Python Version 3.7
 - Mujoco Version 1.50
 - mujoco-py version 1.50.1.68

**Environment**
 - output of: `echo $LD_LIBRARY_PATH` (empty string)
 - output of: `echo $HOME` (my user folder)
 - output of: `echo $USER` (me)

**Additional context**
Add any other context about the problem here.

machinaut, 2018-11-28:
Thank you for the detailed description!

Just to verify:

- can you run the mujoco `simulate` binary (installed with mujoco -- see mujoco documentation for how to use it).  this should check that mujoco can render graphics
- can you run a small python `glfw` program?  this should check that python & glfw work together for rendering

We have not tested Mojave as much as Sierra, but as far as I know downgrading is not possible (or at least very difficult).

xqphd, 2018-11-29:
> Thank you for the detailed description!
>
> Just to verify:
>
> * can you run the mujoco `simulate` binary (installed with mujoco -- see mujoco documentation for how to use it).  this should check that [...]

I have tried the both, and both actually worked fine. 

So I simply ran the `hello.xml` (http://www.mujoco.org/book/index.html#Examples)  on `simulate`, and it worked. 

Then I tried some programmes using `OpenGL` and `glfw` (http://morpheo.inrialpes.fr/~franco/3dgraphics/practical1.html), and they worked well. 

Therefore I think it was not the MuJoCo's problem. Hopefully you could figure out the problems in the future. Thank you.

(I'm not using it in hurry on my laptop, so I can find another device which can run the Gym. Thanks)