openai/mujoco-py #331: gcc error when installing in an Anaconda environment on OS X

angelolovatto, 2018-11-27:
I've successfully installed mujoco-py using pip3 and python 3.7.0
However, I need to install it in an Anaconda environment in order to use python 3.6, as required by the [spinningup](https://spinningup.openai.com/en/latest/user/installation.html#mac-os-x) documentation.
After successfully installing the packages for the base requirements of spinningup, I tried installing mujoco-py using `pip install -U pip install -U 'mujoco-py<1.50.2,>=1.50.1'`.

The installation fails when building the pip package for mujoco-py giving the following fatal error:
`
  /usr/local/bin/gcc-8 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/angelolovatto/anaconda3/envs/spinningup/include -arch x86_64 -I/Users/angelolovatto/anaconda3/envs/spinningup/include -arch x86_64 -DONMAC -Imujoco_py -I/private/var/folders/pj/tpywc3yj6y17s6f964hc7d6c0000gn/T/pip-install-r1g67ly0/mujoco-py/mujoco_py -I/Users/angelolovatto/.mujoco/mjpro150/include -I/Users/angelolovatto/anaconda3/envs/spinningup/lib/python3.6/site-packages/numpy/core/include -I/Users/angelolovatto/anaconda3/envs/spinningup/include/python3.6m -c /private/var/folders/pj/tpywc3yj6y17s6f964hc7d6c0000gn/T/pip-install-r1g67ly0/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/pj/tpywc3yj6y17s6f964hc7d6c0000gn/T/pip-install-r1g67ly0/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder/temp.macosx-10.7-x86_64-3.6/private/var/folders/pj/tpywc3yj6y17s6f964hc7d6c0000gn/T/pip-install-r1g67ly0/mujoco-py/mujoco_py/cymj.o -fopenmp -w
  In file included from /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/syslimits.h:7,
                   from /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/limits.h:34,
                   from /Users/angelolovatto/anaconda3/envs/spinningup/include/python3.6m/Python.h:11,
                   from /private/var/folders/pj/tpywc3yj6y17s6f964hc7d6c0000gn/T/pip-install-r1g67ly0/mujoco-py/mujoco_py/cymj.c:57:
  /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/limits.h:194:15: fatal error: limits.h: No such file or directory
   #include_next <limits.h>  /* recurse down to the real one */
                 ^~~~~~~~~~
  compilation terminated.
  error: command '/usr/local/bin/gcc-8' failed with exit status 1

  ----------------------------------------
  Failed building wheel for mujoco-py
`

This does not happen using homebrew python, which leads me to believe it's an issue involving the Anaconda environment. The files seem to be all there. 

 - OS: macOS 10.14.1
 - Python: Anaconda python 3.6.7
 - Mujoco Version: 1.50

 - output of `echo $LD_LIBRARY_PATH`: empty

angelolovatto, 2018-11-27:
It seems to be a problem with python3.6 in my machine. I verified that I am able to install mujoco-py in Anaconda python3.7 but not python3.6, both Anaconda and Homebrew versions. Could this have something to do with libraries like Cython or boost-python3 on Mac? (which, as far as I know, allow integration of python with c++)

machinaut, 2018-11-28:
@angelolovatto this is a hilarious and terrifying issue.

tl;dr anything compiled using C headers on OSX is affected by apple removing headers.  Install special header package and try again.

Command to fix it is here https://github.com/openai/spinningup/blob/master/travis_setup.sh#L11 (I had to fix this in another one of our projects).

Issue explaining the context for OSX continuous integration builds https://github.com/travis-ci/travis-ci/issues/9640

Let me know if this doesn't help.

closed 2018-11-28 by machinaut

angelolovatto, 2018-12-01:
Unfortunately this didn't help. I tried to install the command-line tools in several ways, however it seems that everything is already installed (maybe something to do with Mojave and Xcode 10.1 ?). I even updated all my homebrew packages to make sure there wasn't anything else missing. 
I'm not sure what to do. If only Tensorflow would update to Python3.7 ... I doubt it's anything to do with mujoco-py and since I won't be able to look much further into this for the next few weeks, I guess it's better to leave this closed for now, unless someone else comes across the same issue. 
Thanks for the help!

Nate711, 2018-12-05:
I'm having the same issue. The command to update the command line tools didn't work.
```
(gym) Nathans-MacBook-Pro-6:mujoco-py nathan$ pip install -e '.[all]'
Obtaining file:///Users/nathan/mujoco-py
  mujoco-py 1.50.1.68 does not provide the extra 'all'
Requirement already satisfied: glfw>=1.4.0 in /anaconda/envs/gym/lib/python3.6/site-packages (from mujoco-py==1.50.1.68) (1.7.0)
Requirement already satisfied: numpy>=1.11 in /anaconda/envs/gym/lib/python3.6/site-packages (from mujoco-py==1.50.1.68) (1.15.4)
Requirement already satisfied: Cython>=0.27.2 in /anaconda/envs/gym/lib/python3.6/site-packages (from mujoco-py==1.50.1.68) (0.29)
Requirement already satisfied: imageio>=2.1.2 in /anaconda/envs/gym/lib/python3.6/site-packages (from mujoco-py==1.50.1.68) (2.4.1)
Requirement already satisfied: cffi>=1.10 in /anaconda/envs/gym/lib/python3.6/site-packages (from mujoco-py==1.50.1.68) (1.11.5)
Requirement already satisfied: lockfile>=0.12.2 in /anaconda/envs/gym/lib/python3.6/site-packages (from mujoco-py==1.50.1.68) (0.12.2)
Requirement already satisfied: pillow in /anaconda/envs/gym/lib/python3.6/site-packages (from imageio>=2.1.2->mujoco-py==1.50.1.68) (5.3.0)
Requirement already satisfied: pycparser in /anaconda/envs/gym/lib/python3.6/site-packages (from cffi>=1.10->mujoco-py==1.50.1.68) (2.19)
Installing collected packages: mujoco-py
  Running setup.py develop for mujoco-py
Successfully installed mujoco-py
(gym) Nathans-MacBook-Pro-6:mujoco-py nathan$ python3
Python 3.6.2 |Continuum Analytics, Inc.| (default, Jul 20 2017, 13:14:59)
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py
Compiling /Users/nathan/mujoco-py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /Users/nathan/mujoco-py/mujoco_py/cymj.pyx
warning: mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
[... more warnings]
warning: mujoco_py/generated/../pxd/mjmodel.pxd:349:8: 'mjDATATYPE_QUAT' redeclared
running build_ext
building 'mujoco_py.cymj' extension
creating /Users/nathan/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder
creating /Users/nathan/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder/temp.macosx-10.7-x86_64-3.6
creating /Users/nathan/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder/temp.macosx-10.7-x86_64-3.6/Users
creating /Users/nathan/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder/temp.macosx-10.7-x86_64-3.6/Users/nathan
creating /Users/nathan/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder/temp.macosx-10.7-x86_64-3.6/Users/nathan/mujoco-py
creating /Users/nathan/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder/temp.macosx-10.7-x86_64-3.6/Users/nathan/mujoco-py/mujoco_py
creating /Users/nathan/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder/temp.macosx-10.7-x86_64-3.6/Users/nathan/mujoco-py/mujoco_py/gl
/usr/local/bin/gcc-8 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I//anaconda/envs/gym/include -arch x86_64 -I//anaconda/envs/gym/include -arch x86_64 -DONMAC -Imujoco_py -I/Users/nathan/mujoco-py/mujoco_py -I/Users/nathan/.mujoco/mjpro150/include -I//anaconda/envs/gym/lib/python3.6/site-packages/numpy/core/include -I//anaconda/envs/gym/include/python3.6m -c /Users/nathan/mujoco-py/mujoco_py/cymj.c -o /Users/nathan/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_macextensionbuilder/temp.macosx-10.7-x86_64-3.6/Users/nathan/mujoco-py/mujoco_py/cymj.o -fopenmp -w
In file included from /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/syslimits.h:7,
                 from /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/limits.h:34,
                 from //anaconda/envs/gym/include/python3.6m/Python.h:11,
                 from /Users/nathan/mujoco-py/mujoco_py/cymj.c:57:
/usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/limits.h:194:15: fatal error: limits.h: No such file or directory
 #include_next <limits.h>  /* recurse down to the real one */
               ^~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "//anaconda/envs/gym/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "//anaconda/envs/gym/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "//anaconda/envs/gym/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "//anaconda/envs/gym/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command '/usr/local/bin/gcc-8' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/nathan/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/Users/nathan/mujoco-py/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/Users/nathan/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
[... cut ...]
    self.build_extension(ext)
  File "//anaconda/envs/gym/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "//anaconda/envs/gym/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "//anaconda/envs/gym/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command '/usr/local/bin/gcc-8' failed with exit status 1
```

PengZhenghao, 2018-12-18:
I encounter the same issue when installing mujoco-py. I am using Python 3.6.7 in my mac with macOS Mojave 10.14.1

The error message in short is 

```
     #include_next <limits.h>  /* recurse down to the real one */
                   ^~~~~~~~~~
    compilation terminated.
    error: command '/usr/local/bin/gcc-7' failed with exit status 1
    
    ----------------------------------------
Command "/Library/Frameworks/Python.framework/Versions/3.6/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-install-75nke5mr/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/tmp/pip-record-89q1ucyf/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/tmp/pip-install-75nke5mr/mujoco-py/

```

The gross error message is:

```
Collecting mujoco-py<1.50.2,>=1.50.1
  Downloading https://pypi.tuna.tsinghua.edu.cn/packages/cf/8c/64e0630b3d450244feef0688d90eab2448631e40ba6bdbd90a70b84898e7/mujoco-py-1.50.1.68.tar.gz (120kB)
    100% |████████████████████████████████| 122kB 2.1MB/s 
Requirement already satisfied, skipping upgrade: glfw>=1.4.0 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mujoco-py<1.50.2,>=1.50.1) (1.7.0)
Requirement already satisfied, skipping upgrade: numpy>=1.11 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mujoco-py<1.50.2,>=1.50.1) (1.15.4)
Requirement already satisfied, skipping upgrade: Cython>=0.27.2 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mujoco-py<1.50.2,>=1.50.1) (0.29.2)
Requirement already satisfied, skipping upgrade: imageio>=2.1.2 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mujoco-py<1.50.2,>=1.50.1) (2.4.1)
Requirement already satisfied, skipping upgrade: cffi>=1.10 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mujoco-py<1.50.2,>=1.50.1) (1.11.5)
Requirement already satisfied, skipping upgrade: lockfile>=0.12.2 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mujoco-py<1.50.2,>=1.50.1) (0.12.2)
Requirement already satisfied, skipping upgrade: pillow in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from imageio>=2.1.2->mujoco-py<1.50.2,>=1.50.1) (5.3.0)
Requirement already satisfied, skipping upgrade: pycparser in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from cffi>=1.10->mujoco-py<1.50.2,>=1.50.1) (2.19)
Installing collected packages: mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /Library/Frameworks/Python.framework/Versions/3.6/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-install-75nke5mr/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/tmp/pip-record-89q1ucyf/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    Removing old mujoco_py cext /private/tmp/pip-install-75nke5mr/mujoco-py/mujoco_py/generated/cymj_1.50.1.68_36_macextensionbuilder_36.so
    Compiling /private/tmp/pip-install-75nke5mr/mujoco-py/mujoco_py/cymj.pyx because it changed.
    [1/1] Cythonizing /private/tmp/pip-install-75nke5mr/mujoco-py/mujoco_py/cymj.pyx
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:121:4: 'mjtJoint' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:127:4: 'mjtGeom' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:149:4: 'mjtCamLight' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:157:4: 'mjtTexture' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:163:4: 'mjtIntegrator' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:168:4: 'mjtCollision' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:174:4: 'mjtCone' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:179:4: 'mjtJacobian' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:185:4: 'mjtSolver' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:191:4: 'mjtImp' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:198:4: 'mjtRef' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:203:4: 'mjtEq' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:211:4: 'mjtWrap' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:220:4: 'mjtTrn' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:230:4: 'mjtDyn' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:237:4: 'mjtGain' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:242:4: 'mjtBias' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:248:4: 'mjtObj' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:274:4: 'mjtConstraint' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:285:4: 'mjtConstraintState' redeclared
    warning: mujoco_py/generated/../pxd/mjmodel.pxd:294:4: 'mjtSensor' redeclared
[... cut ...]
    ----------------------------------------
Command "/Library/Frameworks/Python.framework/Versions/3.6/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-install-75nke5mr/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/tmp/pip-record-89q1ucyf/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/tmp/pip-install-75nke5mr/mujoco-py/
```

thisiscam, 2019-01-01:
Might be late, but https://stackoverflow.com/questions/52509602/cant-compile-c-program-on-a-mac-after-upgrade-to-mojave solved my problem.

Basically, I had to install some auxiliary package that came with Commandlinetools.

Concretely, I did the following:
`sudo open  /Library/Developer/CommandLineTools/Packages/macOS_SDK_headers_for_macOS_10.14.pkg `

quantshah, 2019-02-13:
I had a similar problem on MacOS and it turns out that I should use the Apple supplied `gcc` and not the conda gcc. You can check using

```
which gcc
```

After I removed the conda gcc and updated Xcode, I could install QuTiP and compile stuff. Hope this helps.

miziodel, 2019-06-14:
I received a similar error:
uninstalling conda gcc and using the Xcode gcc solved my issue.
thx!

ReneKat, 2020-02-27:
@miziodel 

Would you be able to help me? I have been chasing this issue all over the internet and nothing has worked so far. 

I am running OSX Mojave 10.14.6 and have just updated Xcode to 11.3.1 trying to solve this gcc compilation error. 

`pip install git+https://github.com/danielfrg/tsne.git`

```
Collecting git+https://github.com/danielfrg/tsne.git
  Cloning https://github.com/danielfrg/tsne.git to /private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-req-build-wu_d_54h
  Running command git clone -q https://github.com/danielfrg/tsne.git /private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-req-build-wu_d_54h
Requirement already satisfied: Cython>=0.19.1 in /Users/Owl/miniconda3/envs/autometa/lib/python3.8/site-packages (from tsne==0.1.8) (0.29.15)
Requirement already satisfied: numpy>=1.7.1 in /Users/Owl/miniconda3/envs/autometa/lib/python3.8/site-packages (from tsne==0.1.8) (1.18.1)
Requirement already satisfied: scipy>=0.12.0 in /Users/Owl/miniconda3/envs/autometa/lib/python3.8/site-packages (from tsne==0.1.8) (1.4.1)
Building wheels for collected packages: tsne
  Building wheel for tsne (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/Owl/miniconda3/envs/autometa/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-req-build-wu_d_54h/setup.py'"'"'; __file__='"'"'/private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-req-build-wu_d_54h/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-wheel-3xolta_u
       cwd: /private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-req-build-wu_d_54h/
  Complete output (29 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.9-x86_64-3.8
  creating build/lib.macosx-10.9-x86_64-3.8/tsne
  copying tsne/_version.py -> build/lib.macosx-10.9-x86_64-3.8/tsne
  copying tsne/__init__.py -> build/lib.macosx-10.9-x86_64-3.8/tsne
  creating build/lib.macosx-10.9-x86_64-3.8/tsne/tests
  copying tsne/tests/__init__.py -> build/lib.macosx-10.9-x86_64-3.8/tsne/tests
  copying tsne/tests/test_seed.py -> build/lib.macosx-10.9-x86_64-3.8/tsne/tests
  copying tsne/tests/test_iris.py -> build/lib.macosx-10.9-x86_64-3.8/tsne/tests
  UPDATING build/lib.macosx-10.9-x86_64-3.8/tsne/_version.py
  set build/lib.macosx-10.9-x86_64-3.8/tsne/_version.py to '0.1.8'
  running build_ext
  building 'bh_sne' extension
  creating build/temp.macosx-10.9-x86_64-3.8
  creating build/temp.macosx-10.9-x86_64-3.8/tsne
  creating build/temp.macosx-10.9-x86_64-3.8/tsne/bh_sne_src
  gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Users/Owl/miniconda3/envs/autometa/include -arch x86_64 -I/Users/Owl/miniconda3/envs/autometa/include -arch x86_64 -I/Users/Owl/miniconda3/envs/autometa/lib/python3.8/site-packages/numpy/core/include -Itsne/bh_sne_src/ -I/Users/Owl/miniconda3/envs/autometa/include/python3.8 -c tsne/bh_sne.cpp -o build/temp.macosx-10.9-x86_64-3.8/tsne/bh_sne.o -I/System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/Headers
  cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
  In file included from /Users/Owl/miniconda3/pkgs/gcc-4.8.2-21/lib/gcc/x86_64-apple-darwin13.0.0/4.8.2/install-tools/include/limits.h:34:0,
                   from /Users/Owl/miniconda3/envs/autometa/include/python3.8/Python.h:11,
                   from tsne/bh_sne.cpp:36:
  /Users/Owl/miniconda3/envs/autometa/lib/gcc/x86_64-apple-darwin11.4.2/4.8.5/include-fixed/syslimits.h:7:25: fatal error: limits.h: No such file or directory
   #include_next <limits.h>
                           ^
  compilation terminated.
  error: command 'gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for tsne
  Running setup.py clean for tsne
Failed to build tsne
Installing collected packages: tsne
    Running setup.py install for tsne ... error
    ERROR: Command errored out with exit status 1:
     command: /Users/Owl/miniconda3/envs/autometa/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-req-build-wu_d_54h/setup.py'"'"'; __file__='"'"'/private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-req-build-wu_d_54h/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-record-5uzqp_sm/install-record.txt --single-version-externally-managed --compile --install-headers /Users/Owl/miniconda3/envs/autometa/include/python3.8/tsne
         cwd: /private/var/folders/5s/r0xdrzyj2dl_43mj6jh0cmzc0000gp/T/pip-req-build-wu_d_54h/
    Complete output (29 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib.macosx-10.9-x86_64-3.8
    creating build/lib.macosx-10.9-x86_64-3.8/tsne
    copying tsne/_version.py -> build/lib.macosx-10.9-x86_64-3.8/tsne
[... cut ...]
    an existing issue about this - open a new one instead.
    
    Original error was: dlopen(/Users/Owl/miniconda3/envs/autometa/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so, 2): Library not loaded: @rpath/libquadmath.0.dylib
      Referenced from: /Users/Owl/miniconda3/envs/autometa/lib/libcblas.3.dylib
      Reason: image not found
    
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```