openai/mujoco-py #332: Failed building wheel

qianwangthu, 2018-12-06:
**Describe the bug**
I'm trying to install with pip but it reports missing file. Thanks in advance for any help. 
 - OS: Ubuntu 16.04 LTS
 - Python Version 3.7.0
 - Mujoco Version 1.50

**Error Messages**
Running setup.py bdist_wheel for mujoco-py ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-9r0h9l5j/mujoco-py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /tmp/tmp_hdf1skmpip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  Removing old mujoco_py cext /tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/generated/cymj_1.50.1.68_35_linuxgpuextensionbuilder_35.so
  Compiling /tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/cymj.pyx because it changed.
  [1/1] Cythonizing /tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/cymj.pyx
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:121:4: 'mjtJoint' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:127:4: 'mjtGeom' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:149:4: 'mjtCamLight' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:157:4: 'mjtTexture' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:163:4: 'mjtIntegrator' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:168:4: 'mjtCollision' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:174:4: 'mjtCone' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:179:4: 'mjtJacobian' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:185:4: 'mjtSolver' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:191:4: 'mjtImp' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:198:4: 'mjtRef' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:203:4: 'mjtEq' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:211:4: 'mjtWrap' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:220:4: 'mjtTrn' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:230:4: 'mjtDyn' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:237:4: 'mjtGain' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:242:4: 'mjtBias' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:248:4: 'mjtObj' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:274:4: 'mjtConstraint' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:285:4: 'mjtConstraintState' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:294:4: 'mjtSensor' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:338:4: 'mjtStage' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:345:4: 'mjtDataType' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:99:4: 'mjtDisableBit' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:114:4: 'mjtEnableBit' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:121:4: 'mjtJoint' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:127:4: 'mjtGeom' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:149:4: 'mjtCamLight' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:157:4: 'mjtTexture' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:163:4: 'mjtIntegrator' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:168:4: 'mjtCollision' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:174:4: 'mjtCone' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:179:4: 'mjtJacobian' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:185:4: 'mjtSolver' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:191:4: 'mjtImp' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:198:4: 'mjtRef' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:203:4: 'mjtEq' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:211:4: 'mjtWrap' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:220:4: 'mjtTrn' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:230:4: 'mjtDyn' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:237:4: 'mjtGain' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:242:4: 'mjtBias' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:248:4: 'mjtObj' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:274:4: 'mjtConstraint' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:285:4: 'mjtConstraintState' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:294:4: 'mjtSensor' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:338:4: 'mjtStage' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:345:4: 'mjtDataType' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:75:4: 'mjtNum' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:77:10: 'mjPI' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:78:10: 'mjMAXVAL' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:79:10: 'mjMINMU' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:80:10: 'mjMINIMP' redeclared
  warning: mujoco_py/generated/../pxd/mjmodel.pxd:81:10: 'mjMAXIMP' redeclared
[... 366 lines cut ...]
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for mujoco-py
  Running setup.py clean for mujoco-py
Failed to build mujoco-py
Installing collected packages: glfw, numpy, Cython, pillow, imageio, pycparser, cffi, lockfile, mujoco-py
  Running setup.py install for mujoco-py ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-9r0h9l5j/mujoco-py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-v3ofip5d-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    Removing old mujoco_py cext /tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/generated/cymj_1.50.1.68_35_linuxgpuextensionbuilder_35.so
    running build_ext
    building 'mujoco_py.cymj' extension
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py -I/home/duinduin/.mujoco/mjpro150/include -I/home/duinduin/.local/lib/python3.5/site-packages/numpy/core/include -I/tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/vendor/egl -I/usr/include/python3.5m -c /tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/cymj.c -o /tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_35_linuxgpuextensionbuilder/temp.linux-x86_64-3.5/tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py -I/home/duinduin/.mujoco/mjpro150/include -I/home/duinduin/.local/lib/python3.5/site-packages/numpy/core/include -I/tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/vendor/egl -I/usr/include/python3.5m -c /tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/gl/eglshim.c -o /tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_35_linuxgpuextensionbuilder/temp.linux-x86_64-3.5/tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/gl/eglshim.o -fopenmp -w
    /tmp/pip-build-9r0h9l5j/mujoco-py/mujoco_py/gl/eglshim.c:4:21: fatal error: GL/glew.h: No such file or directory
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-9r0h9l5j/mujoco-py/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-v3ofip5d-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-9r0h9l5j/mujoco-py/

**Desktop (please complete the following information):**

qianwangthu, 2018-12-06:
It has been resolved by install libglew-dev.

closed 2018-12-06 by qianwangthu

drozzy, 2019-02-11:
What? How did you resolve it (did you use brew to install it - cause I can't)?

It's still failing for me....

sheilaschoepp, 2020-11-12:
I had this same problem.  Took me 12 hours to figure it out.  I hope that this helps others out there!  

do the following two commands right after one another:
apt install libgl1-mesa-glx --reinstall
ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1 /usr/lib/x86_64-linux-gnu/libGL.so

you may get an error saying the file already exists.  if so, do the following (in addition to the above):
rm /usr/lib/x86_64-linux-gnu/libGL.so
ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1 /usr/lib/x86_64-linux-gnu/libGL.so

Basically, I found a source that said the symbolic link could be pointing to the wrong thing.

Once I did this, I was able to install mujoco-py and then gym[all]