openai/mujoco-py #335: RuntimeError: Window rendering not supported

giorgionicola, 2018-12-20:
**Describe the bug**
I am working with the library for keras-rl (https://github.com/keras-rl/keras-rl) when I try to run any algorithm workin on a mujoco environment the visualization fails giving me an error in the mujoco-py package:

`File "ddpg_mujoco.py", line 76, in <module>`
    `agent.test(env, nb_episodes=5, visualize=True, nb_max_episode_steps=200)`
`File "/media/DATA/anaconda3/envs/kerasrl/lib/python3.6/site-packages/rl/core.py", line 353, in test
    `callbacks.on_action_end(action)`
  File "/media/DATA/anaconda3/envs/kerasrl/lib/python3.6/site-packages/rl/callbacks.py", line 101, in on_action_end`
    `callback.on_action_end(action, logs=logs)`
  `File "/media/DATA/anaconda3/envs/kerasrl/lib/python3.6/site-packages/rl/callbacks.py", line 366, in on_action_end`
    `self.env.render(mode='human')`
  `File "/media/DATA/anaconda3/envs/kerasrl/lib/python3.6/site-packages/gym/core.py", line 275, in render`
    `return self.env.render(mode, **kwargs)`
  `File "/media/DATA/anaconda3/envs/kerasrl/lib/python3.6/site-packages/gym/core.py", line 275, in render`
    `return self.env.render(mode, **kwargs)`
 ` File "/media/DATA/anaconda3/envs/kerasrl/lib/python3.6/site-packages/gym/envs/mujoco/mujoco_env.py", line 118, in render`
   ` self._get_viewer(mode).render()`
 ` File "/media/DATA/anaconda3/envs/kerasrl/lib/python3.6/site-packages/gym/envs/mujoco/mujoco_env.py", line 130, in _get_viewer`
   ` self.viewer = mujoco_py.MjViewer(self.sim)`
  `File "/media/DATA/anaconda3/envs/kerasrl/lib/python3.6/site-packages/mujoco-py/mujoco_py/mjviewer.py", line 133, in __init__`
    `super().__init__(sim)`
 ` File "/media/DATA/anaconda3/envs/kerasrl/lib/python3.6/site-packages/mujoco-py/mujoco_py/mjviewer.py", line 26, in __init__`
   ` super().__init__(sim)`
  `File "mjrendercontext.pyx", line 278, in mujoco_py.cymj.MjRenderContextWindow.__init__`
 ` File "mjrendercontext.pyx", line 66, in mujoco_py.cymj.MjRenderContext.__init__`
  `File "mjrendercontext.pyx", line 87, in mujoco_py.cymj.MjRenderContext._set_mujoco_buffers`
`RuntimeError: Window rendering not supported`

It is not clear to me the meaning of the error. Any help would be appreciated

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04
 - Python Version 3.6.7
 - Mujoco Version 1.50
 - mujoco-py version 1.50.1.68

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
`/usr/local/cuda/extras/CUPTI/lib64:/home/tartaglia/Libraries/mujoco/mjpro150/bin:/usr/lib/nvidia-384:/home/tartaglia/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/usr/local/cuda/lib64:/home/tartaglia/gpu_voxels/export/lib`

 - output of: `echo $HOME`
`/home/tartaglia`

 - output of: `echo $USER`
`tartaglia`

vitchyr, 2019-03-18:
I'm not sure what causes this, but a work-around that works for me is to use `env.sim.render(width=w, height=h, camera_name=...)`

huangjiancong1, 2020-04-23:
> **Describe the bug**
> I am working with the library for keras-rl (https://github.com/keras-rl/keras-rl) when I try to run any algorithm workin on a mujoco environment the visualization fails giving [...]

Hi，@giorgionicola， I have same issue like your, have you fixed it?

huangjiancong1, 2020-04-23:
I fixed it in by [https://github.com/openai/mujoco-py/issues/187#issuecomment-618321489](url).