openai/mujoco-py #336: fatal error: X11/Xlib.h: No such file or directory when import mujoco_py

huiwenzhang, 2018-12-28:
**Describe the bug**
When import the module, compile failed with erro
> >>> import mujoco_py
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/huiwen/anaconda3/envs/gail/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py -I/home/huiwen/.mujoco/mjpro150/include -I/home/huiwen/anaconda3/envs/gail/lib/python3.6/site-packages/numpy/core/include -I/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/vendor/egl -I/home/huiwen/anaconda3/envs/gail/include/python3.6m -c /home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/cymj.c -o /home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/huiwen/anaconda3/envs/gail/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py -I/home/huiwen/.mujoco/mjpro150/include -I/home/huiwen/anaconda3/envs/gail/lib/python3.6/site-packages/numpy/core/include -I/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/vendor/egl -I/home/huiwen/anaconda3/envs/gail/include/python3.6m -c /home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/gl/eglshim.c -o /home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/generated/_pyxbld_1.50.1.68_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/gl/eglshim.o -fopenmp -w
In file included from /home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/gl/egl.h:39:0,
                 from /home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/gl/eglshim.c:2:
/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/gl/eglplatform.h:99:10: fatal error: X11/Xlib.h: No such file or directory
 #include <X11/Xlib.h>
          ^~~~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/builder.py", line 290, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/builder.py", line 243, in _build_impl
    dist.run_commands()
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/huiwen/paper/learn-from-video/mujoco-py/mujoco_py/builder.py", line 143, in build_extensions
    build_ext.build_extensions(self)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
    self.build_extension(ext)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/huiwen/anaconda3/envs/gail/lib/python3.6/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'gcc' failed with exit status 1

**Error Messages**
fatal error: X11/Xlib.h: No such file or directory.

**Desktop (please complete the following information):**
 - OS: [Ubutnu 16.04]
 - Python Version [3.6.4]
 - Mujoco Version [ 1.50]
 - mujoco-py version [e.g. 1.50.1.68]
 - GCC version [7.2.0]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
 - output of: `echo $HOME`
 - output of: `echo $USER`

**Additional context**
I find the Xlib.h file with `locate' command and get:
> /home/huiwen/anaconda3/pkgs/xorg-libx11-1.6.5-0/include/X11/Xlib.h
/home/xiaoning/anaconda3/envs/py35/include/X11/Xlib.h
/home/xiaoning/anaconda3/pkgs/xorg-libx11-1.6.6-h470a237_0/include/X11/Xlib.h
/usr/include/X11/Xlib.h
/usr/lib/x86_64-linux-gnu/perl5/5.22/Tk/pTk/Xlib.h
It does exist. However, I have no idea why the system can't find it.

sxwgit, 2019-01-28:
I get a same issue 

> eglplatform.h:99:10: fatal error: X11/Xlib.h: No such file or directory
and i had tried your answer,unfortunately，still get this problem
> songxw@songxw-pc:~/sysroot/usr/include$ pwd
> /home/songxw/sysroot/usr/include
> songxw@songxw-pc:~/sysroot/usr/include$ ls
> X11  Xlib.h

i copy X11 and Xlib.h to /sysroot/usr/include ,would you mind tell me more details how you solve this problem,
Best regards

HACLANG, 2021-11-25:
I just rerun the following command to fix the problem

`conda install -c conda-forge --yes --file requirements.txt`
`conda install -c conda-forge --yes --file requirements.dev.txt``

G0K0URURI, 2023-01-14:
I fix the problem by installing x11:
`sudo apt-get install libghc-x11-dev`

G0K0URURI, 2023-01-14:
> I get a same issue
>
>
> and i had tried your answer,unfortunately，still get this problem
>
>
> i copy X11 and Xlib.h to /sysroot/usr/include ,would you mind tell me more details how you solve this problem, [...]
I fix the problem by installing x11:
`sudo apt-get install libghc-x11-dev`