openai/mujoco-py #337: can't import mujoco_py

bluewhitep, 2019-01-04:
Python 3.6.7 (default, Oct 22 2018, 11:32:17) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco_py

You appear to be missing MuJoCo.  We expected to find the file here: ${HOME}/.mujoco/mjpro150

This package only provides python bindings, the library must be installed separately.

Please follow the instructions on the README to install MuJoCo

    https://github.com/openai/mujoco-py#install-mujoco

Which can be downloaded from the website

    https://www.roboti.us/index.html

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/builder.py", line 502, in <module>
    mjpro_path, key_path = discover_mujoco()
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/utils.py", line 93, in discover_mujoco
    raise Exception(message)
Exception: 
You appear to be missing MuJoCo.  We expected to find the file here: ${HOME}/.mujoco/mjpro150

This package only provides python bindings, the library must be installed separately.

Please follow the instructions on the README to install MuJoCo

    https://github.com/openai/mujoco-py#install-mujoco

Which can be downloaded from the website

    https://www.roboti.us/index.html

>>> exit()
bluewhite@Ubuntu-server:~/Documents/workspace/openai/mujoco-py$ la ${HOME}/.mujoco/mjpro150
bin  doc  include  model  sample

closed 2019-01-04 by bluewhitep

DesparadoW, 2020-05-29:
Have you resolve this problem? I just meet that now, can't find any solution.

amirhossein-nhn, 2021-08-24:
Hi,
I have the same problem. Did anyone find anything helpful?