openai/mujoco-py #338: can't import mujoco_py

bluewhitep, 2019-01-04:
Hi, I try import mujoco_py,  but it return error message

import mujoco_py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/builder.py", line 105, in load_cython_ext
    cext_so_path = builder.build()
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/builder.py", line 220, in build
    built_so_file_path = self._build_impl()
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/builder.py", line 272, in _build_impl
    so_file_path = super()._build_impl()
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/builder.py", line 233, in _build_impl
    dist.ext_modules = cythonize([self.extension])
  File "/home/bluewhite/.local/lib/python3.6/site-packages/Cython/Build/Dependencies.py", line 958, in cythonize
    ctx = c_options.create_context()
  File "/home/bluewhite/.local/lib/python3.6/site-packages/Cython/Compiler/Main.py", line 595, in create_context
    self.cplus, self.language_level, options=self)
  File "/home/bluewhite/.local/lib/python3.6/site-packages/Cython/Compiler/Main.py", line 78, in __init__
    from . import Builtin, CythonScope
  File "/home/bluewhite/.local/lib/python3.6/site-packages/Cython/Compiler/CythonScope.py", line 5, in <module>
    from .UtilityCode import CythonUtilityCode
  File "/home/bluewhite/.local/lib/python3.6/site-packages/Cython/Compiler/UtilityCode.py", line 3, in <module>
    from .TreeFragment import parse_from_strings, StringParseContext
  File "/home/bluewhite/.local/lib/python3.6/site-packages/Cython/Compiler/TreeFragment.py", line 17, in <module>
    from .Visitor import VisitorTransform
  File "Cython/Compiler/Visitor.py", line 17, in init Cython.Compiler.Visitor
  File "/home/bluewhite/.local/lib/python3.6/site-packages/Cython/Compiler/ExprNodes.py", line 46, in <module>
    from .Pythran import (to_pythran, is_pythran_supported_type, is_pythran_supported_operation_type,
ImportError: cannot import name 'pythran_is_numpy_func_supported'

I tried downgrading to cython 0.28.5
I get new error message

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/hdd/workspace/openai/mujoco-py/mujoco_py/builder.py", line 17, in <module>
    from Cython.Build import cythonize
ModuleNotFoundError: No module named 'Cython.Build'

bluewhitep, 2019-01-04:
new error message：

>>> import mujoco_py
ERROR: Could not read activation key

Press Enter to exit ...

bluewhite@Ubuntu-server:~$ find ~/.mujoco | grep 'mjkey'
/home/bluewhite/.mujoco/mjpro150/bin/mjkey.txt
/home/bluewhite/.mujoco/mjkey.txt

HareshMiriyala, 2019-01-05:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'mujoco_py'

HarryRhythm, 2019-06-15:
I met the same problem, have you solved yet

bluewhitep, 2019-06-15:
No，I gave up

jdzied, 2020-01-23:
You need to set the environment variable directly to the .txt file - not the folder where it is located.
I.e. in .bashrc:
export MUJOCO_PY_MJKEY_PATH="$HOME/.mujoco/mujoco200/bin/mjkey.txt"
Hope this helps?

Settheworldonfireiii, 2023-03-27:
> You need to set the environment variable directly to the .txt file - not the folder where it is located. I.e. in .bashrc: export MUJOCO_PY_MJKEY_PATH="$HOME/.mujoco/mujoco200/bin/mjkey.txt" Hope this helps?

it did not.