openai/mujoco-py #345: cython recompiles every time

richardrl, 2019-01-25:
**Describe the bug**
Compiling /env/lib/python3.6/site-packages/mujoco_py-1.50.1.68-py3.6.egg/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /env/lib/python3.6/site-packages/mujoco_py-1.50.1.68-py3.6.egg/mujoco_py/cymj.pyx
...

Every time the built docker file runs, it recompiles the cython.
**To Reproduce**
Steps to reproduce the behavior.
Run any docker image with mujoco-py installed and import mujoco_py
**Expected behavior**
It should not recythonize after being built - it takes a long time to recythonize
**Error Messages**
Including more/longer error messages gives us more information to work with.

**Desktop (please complete the following information):**
 - OS: MacOS/Ubuntu/Docker
 - Python Version 3
 - Mujoco Version1.5
 - mujoco-py version: latest

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
 - output of: `echo $HOME`
 - output of: `echo $USER`

**Additional context**
Add any other context about the problem here.

BTickell, 2019-03-19:
The first time mujoco_py is imported the cythonizing occurs. If you are not commiting the modified container then the starting image state remains unchanged and launching the image will not have the compiled mujoco_py.

im-Kitsch, 2023-02-27:
I also have this problem in cluster. 

When I try the program at login-node, it doesn't matter. But when I submit it via slurm. It cannot run at compute node. It runs cythoning with information "Compiling ...../mujoco_py/cymj.pyx because it changed."