openai/mujoco-py #348: collect2: error: ld returned 1 exit status

sxwgit, 2019-01-30:
**Describe the bug**
When running the behavior_clone.py on the conda environment，i get the following error

> collect2: error: ld returned 1 exit status
> Traceback (most recent call last):
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/unixccompiler.py", line 197, in link
>     self.spawn(linker + ld_args)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
>     spawn(cmd, dry_run=self.dry_run)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/spawn.py", line 36, in spawn
>     _spawn_posix(cmd, search_path, dry_run=dry_run)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
>     % (cmd, exit_status))
> distutils.errors.DistutilsExecError: command 'x86_64-conda_cos6-linux-gnu-gcc' failed with exit status 1
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "run_mujoco.py", line 245, in <module>
>     main(args)
>   File "run_mujoco.py", line 79, in main
>     env = gym.make(args.env_id)
>   File "/home/songxw/git/gym/gym/envs/registration.py", line 170, in make
>     return registry.make(id)
>   File "/home/songxw/git/gym/gym/envs/registration.py", line 119, in make
>     env = spec.make()
>   File "/home/songxw/git/gym/gym/envs/registration.py", line 85, in make
>     cls = load(self._entry_point)
>   File "/home/songxw/git/gym/gym/envs/registration.py", line 14, in load
>     result = entry_point.resolve()
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2331, in resolve
>     module = __import__(self.module_name, fromlist=['__name__'], level=0)
>   File "/home/songxw/git/gym/gym/envs/mujoco/__init__.py", line 1, in <module>
>     from gym.envs.mujoco.mujoco_env import MujocoEnv
>   File "/home/songxw/git/gym/gym/envs/mujoco/mujoco_env.py", line 11, in <module>
>     import mujoco_py
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/mujoco_py/__init__.py", line 3, in <module>
>     from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/mujoco_py/builder.py", line 503, in <module>
>     cymj = load_cython_ext(mjpro_path)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/mujoco_py/builder.py", line 105, in load_cython_ext
>     cext_so_path = builder.build()
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/mujoco_py/builder.py", line 220, in build
>     built_so_file_path = self._build_impl()
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/mujoco_py/builder.py", line 290, in _build_impl
>     so_file_path = super()._build_impl()
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/mujoco_py/builder.py", line 243, in _build_impl
>     dist.run_commands()
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/dist.py", line 955, in run_commands
>     self.run_command(cmd)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
>     _build_ext.build_ext.run(self)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/command/build_ext.py", line 339, in run
>     self.build_extensions()
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/mujoco_py/builder.py", line 143, in build_extensions
>     build_ext.build_extensions(self)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/site-packages/Cython/Distutils/old_build_ext.py", line 194, in build_extensions
>     self.build_extension(ext)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/command/build_ext.py", line 558, in build_extension
>     target_lang=language)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/ccompiler.py", line 717, in link_shared_object
>     extra_preargs, extra_postargs, build_temp, target_lang)
>   File "/home/songxw/anaconda3/envs/mujoco-py/lib/python3.5/distutils/unixccompiler.py", line 199, in link
>     raise LinkError(msg)
> distutils.errors.LinkError: command 'x86_64-conda_cos6-linux-gnu-gcc' failed with exit status 1

in this [issue](https://github.com/openai/baselines/issues/803) , i got two problem and i solved the first problem.But i dont  have any ideas about how to solve this problem, i guess its a gcc problem not a misheader files problem? would you mind tell how to solve it?
Best regards,

my conda info is 

>      active environment : mujoco-py
>     active env location : /home/songxw/anaconda3/envs/mujoco-py
>             shell level : 2
>        user config file : /home/songxw/.condarc
>  populated config files : 
>           conda version : 4.5.12
>     conda-build version : 3.17.6
>          python version : 3.7.1.final.0
>        base environment : /home/songxw/anaconda3  (writable)
>            channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
>                           https://repo.anaconda.com/pkgs/main/noarch
>                           https://repo.anaconda.com/pkgs/free/linux-64
>                           https://repo.anaconda.com/pkgs/free/noarch
>                           https://repo.anaconda.com/pkgs/r/linux-64
>                           https://repo.anaconda.com/pkgs/r/noarch
>                           https://repo.anaconda.com/pkgs/pro/linux-64
>                           https://repo.anaconda.com/pkgs/pro/noarch
>           package cache : /home/songxw/anaconda3/pkgs
>                           /home/songxw/.conda/pkgs
>        envs directories : /home/songxw/anaconda3/envs
>                           /home/songxw/.conda/envs
>                platform : linux-64
>              user-agent : conda/4.5.12 requests/2.21.0 CPython/3.7.1 Linux/4.15.0-29-generic ubuntu/16.04 glibc/2.23
>                 UID:GID : 1000:1000
>              netrc file : None
>            offline mode : False
> 

ubuntu 16.04
python version 3.5.6
conda version 4.5.12
mujoco mjpro150

closed 2019-02-05 by sxwgit

atishdixit16, 2019-04-10:
Same issue here.
 @sxwgit, How did you solve this issue?

sxwgit, 2019-04-10:
In fact, I don't know why this problem occurs，sorry...
But I solved this problem by reinstalling mujoco and relevant configuration.
I also used this [method ](https://github.com/openai/baselines/issues/803) .