openai/mujoco-py #350: Install failed on OS X cause Cython

XuSen123, 2019-02-01:
Hi there,
Below are logs and error messages I get when attempting the pip installation of mujoco-py.
I'm running OS X 10.12.6, in a conda env with python 3.6

```
from . import Parsing
      File "Cython/Compiler/Scanning.pxd", line 18, in init Cython.Compiler.Parsing (/Users/ilan/minonda/conda-bld/cython_1482522032589/work/Cython-0.25.2/Cython/Compiler/Parsing.c:67386)
    ValueError: Cython.Compiler.Scanning.CompileTimeScope has the wrong size, try recompiling. Expected 32, got 40

    ----------------------------------------
Command "/Users/SENHAI/anaconda/bin/python -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-build-14uhn73b/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-gns53ua8-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/tmp/pip-build-14uhn73b/mujoco-py/
```

XuSen123, 2019-02-02:
it seems that mujoco_py cannot support python3.6 well. i build it in python3.5 succesfullt.

JudyYe, 2019-02-19:
Is there a solution just for python 3.6 on MacOSX?  Other packages require python >=3.6

JudyYe, 2019-09-06:
Reinstalling cython resolve the problem for me!!
```
conda install -c anaconda cython 
```