openai/mujoco-py #356: Errors during pip install mujoco_py

drozzy, 2019-02-11:
Encountering this error when installing on Mac OSX:

`pip install mujoco-py`

```
In file included from /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/syslimits.h:7,
                   from /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/limits.h:34,
                   from /Users/andriy/anaconda3/envs/surreal/include/python3.7m/Python.h:11,
                   from /private/var/folders/qb/fvg5ts7n4y1gplffdfdmmrfm0000gn/T/pip-install-i11b6svg/mujoco-py/mujoco_py/cymj.c:57:
  /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/limits.h:194:15: fatal error: limits.h: No such file or directory
   #include_next <limits.h>  /* recurse down to the real one */
                 ^~~~~~~~~~
  compilation terminated.
  error: command '/usr/local/bin/gcc-8' failed with exit status 1

Failed building wheel for mujoco-py

In file included from /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/syslimits.h:7,
                     from /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/limits.h:34,
                     from /Users/andriy/anaconda3/envs/surreal/include/python3.7m/Python.h:11,
                     from /private/var/folders/qb/fvg5ts7n4y1gplffdfdmmrfm0000gn/T/pip-install-i11b6svg/mujoco-py/mujoco_py/cymj.c:57:
    /usr/local/Cellar/gcc/8.2.0/lib/gcc/8/gcc/x86_64-apple-darwin17.7.0/8.2.0/include-fixed/limits.h:194:15: fatal error: limits.h: No such file or directory
     #include_next <limits.h>  /* recurse down to the real one */
                   ^~~~~~~~~~
    compilation terminated.
    error: command '/usr/local/bin/gcc-8' failed with exit status 1
    
    ----------------------------------------
Command "/Users/andriy/anaconda3/envs/surreal/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/qb/fvg5ts7n4y1gplffdfdmmrfm0000gn/T/pip-install-i11b6svg/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/qb/fvg5ts7n4y1gplffdfdmmrfm0000gn/T/pip-record-zo8q9egp/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/qb/fvg5ts7n4y1gplffdfdmmrfm0000gn/T/pip-install-i11b6svg/mujoco-py/

```

drozzy, 2019-02-11:
I've placed the folder `mjpro150` into `~/.mujoco` and also placed the `mjkey.txt` in there.

episodeyang, 2019-02-21:
I needed to 

1. brew uninstall `gcc`
> notifies that there is still a `gcc@7` left in the system.
2. brew uninstall `gcc@7`
3. brew install `gcc@8`
4. pip install `mujoco-py`.

sungsulim, 2019-04-16:
For me it was somehow using gcc@6 and had to do
1. brew uninstall gcc
2. brew uninstall gcc@6
3. brew install gcc@8
4. pip3 install mujoco-py

marjanin, 2019-06-17:
It still did not work for me.
I am trying to run it on a mac and the full error is:

 ERROR: Complete output from command /Users/alimarjaninejad/.pyenv/versions/3.5.2/bin/python3.5 -u -c 'import setuptools, tokenize;__file__='"'"'/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-record-y5h4ccie/install-record.txt --single-version-externally-managed --compile:
    ERROR: running install
    running build
    Removing old mujoco_py cext /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/cymj_2.0.2.2_35_macextensionbuilder_35.so
    Compiling /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/cymj.pyx because it depends on /Users/alimarjaninejad/.pyenv/versions/3.5.2/lib/python3.5/site-packages/Cython/Includes/numpy/__init__.pxd.
    [1/1] Cythonizing /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/cymj.pyx
    running build_ext
    building 'mujoco_py.cymj' extension
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var/folders
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var/folders/s6
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py
    creating /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/gl
    /usr/local/bin/gcc-8 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -DONMAC -Imujoco_py -I/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py -I/Users/alimarjaninejad/.mujoco/mujoco200/include -I/Users/alimarjaninejad/.pyenv/versions/3.5.2/lib/python3.5/site-packages/numpy/core/include -I/Users/alimarjaninejad/.pyenv/versions/3.5.2/include/python3.5m -c /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/cymj.c -o /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/generated/_pyxbld_2.0.2.2_35_macextensionbuilder/temp.macosx-10.14-x86_64-3.5/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py/cymj.o -fopenmp -w
    /usr/local/bin/gcc-8 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -DONMAC -Imujoco_py -I/private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/mujoco_py -I/Users/alimarjaninejad/.mujoco/mujoco200/include -I/Users/alimarjaninejad/.pyenv/versions/3.5.2/lib/python3.5/site-packages/numpy/core/include -I/Users/alimarjaninejad/.pyenv/versions/3.5.2/include/py [... cut ...] =getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-record-y5h4ccie/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/s6/0_d9s9257qj2dwhc9nfpv0_c0000gn/T/pip-install-v9u6jsps/mujoco-py/

has anyone faced the same problem and found a solution for it?

marjanin, 2019-06-17:
It worked for me once I installed the pkg mentioned here:
https://stackoverflow.com/questions/51274905/brew-install-on-mojave

RyanRizzo96, 2019-10-11:
I'm still having an issue with Mac OS Catalina 10.15:

```
  error: command '/usr/local/bin/gcc-6' failed with exit status 1
    ----------------------------------------
ERROR: Command "/usr/local/opt/python/bin/python3.7 -u -c 'import setuptools, tokenize;__file__='"'"'/private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-9cr4hkj6/mujoco-py/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-record-yj96orw2/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/q7/q_59j36d5n31209r5c01mx2w0000gp/T/pip-install-9cr4hkj6/mujoco-py/

Anyone else experiencing this?
```