openai/mujoco-py #357: Offcreen framebuffer is not complete

harjatinsingh, 2019-02-13:
Using offscreen rendering throws this error message with Tesla P100 GPU 
ERROR: Offscreen framebuffer is not complete, error 0x8cdd.

However the same code works with 1080 Ti and Titan X GPUs. Is there something I need to modify. I am using mujoco1.5 with Ananconda Python 3.6.5.

Gaoee, 2019-04-19:
I have similar error too.

ajabri, 2019-04-26:
Bump on this. Did you guys figure this issue out?

yusukeurakami, 2019-05-02:
I have it too. In my case, it happens after I started using pytorch.multiprocess. Mujoco 1.55, Mujoco-py 1.5 Titan X GPU. Ubuntu 16.04

BTickell, 2019-05-29:
@yusukeurakami any luck?

yusukeurakami, 2019-05-29:
@BTickell 
So, in my case, this problem happened because of the off-screen framebuffer became full by opening too many Mjviewers. Every iteration, I was creating the new `MujocoEnv` and calling `MujocoEnv._get_viewer()` for rendering purpose. However, I realized that the viewer will not be deleted even I deleted the env itself and eventually it will full up my memory space. 
Therefore, I stopped calling the _get_viewer() function and use Unity to render the image instead of Mujoco renderer.

BTickell, 2019-05-29:
Interesting, I am actually getting this error when I STOP naively generating new render contexts. In my case I am using  offscreen rendering and when creating a new simulator I first access the original doing

```
rc = old_sim.render_contexts[0]
new_sim = mujoco_py.MjSim(model)
sim.add_render_context(rc)
rc.update_sim(new_sim)
```
If I remove this process the error goes away...

tkelestemur, 2019-07-19:
I'm getting the same issue with the following:
```        
    def __init__(self, render=False):

        self.model = mj.load_model_from_path(MODEL_PATH)
        self.sim = mj.MjSim(self.model)
        if render:
            self.viewer = mj.MjViewer(self.sim)
        else:
            self.depth_viewer = mj.MjRenderContextOffscreen(self.sim)
```
If the `render` is `False` it imideatly throws the same error. I don't even need to call or render or read_pixel functions. I'm on Ubuntu 18.04 with Titan RTX.