openai/mujoco-py #358: Setting up MuJoCo on Google Colaboratory

Akella17, 2019-02-16:
The ipython notebook can be found in this link: https://drive.google.com/file/d/1L-SzAYSNLruiJttNrnSk7W2vex2e9yev/view?usp=sharing

mokeam, 2019-02-18:
hi, have you been able to fix this?

Akella17, 2019-02-18:
nope

Akella17, 2019-02-20:
I think I found the issue. MuJoCo's license is based on the activation keys generated from the computer and OS (limited to 3 registrations per year). This practically makes it impossible to use MuJoCo with Colabs.
Hope this helps!!

closed 2019-02-20 by Akella17

10jinn01, 2020-07-02:
Get an institutional license.

KodinGuy, 2021-02-15:
Another solution would be to setup a remote server between colab and a licensed pc.

KodinGuy, 2021-02-15:
Using a rpc server and client.

BuildingAtom, 2021-12-08:
In case anyone else needs mujoco-py (say for some OpenAI Gym environment), I did get it working with latest version of MuJoCo since it doesn't need a hardware key. I put the code block I wrangled together into a gist here: https://gist.github.com/BuildingAtom/3119ac9c595324c8001a7454f23bf8c8

I had to use MuJoCo on another project and took the time to figure this out issue for Colab. Hopefully this helps some others in the same boat!