openai/mujoco-py #359: Unable to use Humanoid v2

mokeam, 2019-02-18:
**I have tried setting the path in .bashrc and .profile but it's still throwing the same error. Please see the output below**

>>> import gym
>>> env = gym.make('Humanoid-v2')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 171, in make
    return registry.make(id, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 123, in make
    env = spec.make(**kwargs)
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 86, in make
    cls = load(self._entry_point)
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 14, in load
    result = entry_point.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/mujoco/__init__.py", line 1, in <module>
    from gym.envs.mujoco.mujoco_env import MujocoEnv
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/mujoco/mujoco_env.py", line 11, in <module>
    import mujoco_py
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.68-py3.5.egg/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.68-py3.5.egg/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.68-py3.5.egg/mujoco_py/builder.py", line 69, in load_cython_ext
    _ensure_set_env_var("LD_LIBRARY_PATH", lib_path)
  File "/usr/local/lib/python3.5/dist-packages/mujoco_py-1.50.1.68-py3.5.egg/mujoco_py/builder.py", line 118, in _ensure_set_env_var
    var_name, var_name, lib_path))
Exception: 
Missing path to your environment variable. 
Current values LD_LIBRARY_PATH=
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/mokeam/.mujoco/mjpro150/bin

jaggernaut007, 2019-03-22:
This is usually because the processor is an old one.

mokeam, 2019-03-22:
Thank you,i later figured out that that was the issue.

closed 2019-03-22 by mokeam