openai/mujoco-py #362: Can someone confirm Mujoco_py will build/install on WSL (Windows Subsystem for Linux)?

Christopheraburns, 2019-02-27:
I am running Ubuntu 18.04 on Windows 10.

I am unable to install mujoco from the instructions.  I first got the error in this thread:

https://github.com/openai/mujoco-py/issues/90

I used apt-file to correct but then received an error (I did not save it) related to the compare_compiler "-lGL".

I updated to the gcc_linux_64 compile and I am back to the GL/osmesa.h missing error (the issue linked above)!

according apt-file  libosmesa6-dev is installed and the osmesa.h file is in /usr/include/GL/ directory.

Looking for someone to confirm that mujoco DOES install in WSL before I put any more time into this.

TIA.

fleeb24, 2019-03-02:
I have mujoco-py and dm_control running on WSL (running Ubuntu 18.04 on Windows 10). I set things up a while ago and I remember having some trouble with osmesa or Open GL or something. Unfortunately, I don't remember exactly what ended up working. 

I managed to figure out how to get it working, so it shouldn't require more than installing stuff with apt and pip (using anaconda), and it shouldn't be too difficult :)
At some point I also had the visualization working using VcXsrv, however it's not working for me at the moment. I have since switched to dm_control which is working much better for me.

Sorry I can't be more help. Anyway, at least I can confirm that mujoco-py does work on WSL.

bluephoton, 2019-03-03:
Thanks @fleeb24 , dm_control took 5 mins to install on windows and was able to render as well! mujoco-py is dropping windows support it seems!

sam-the-programmer, 2022-09-08:
https://stackoverflow.com/questions/46182673/mujoco-linux-package-installation-error-exit-status-1 this has some useful information. I am just trying to get it to work now