openai/mujoco-py #364: gym.make() is frozen, takes infinite time to run

HareshMiriyala, 2019-02-28:
**Describe the bug**
when i run the following code, 
`env = gym.make('InvertedPendulum-v2')`  ,the lines takes infinite time to run. It doesn't return a value. 

**Error Messages**
No error messages. It just doesn't seem to make the environment. 

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04
 - Python Version Python 3.6 Conda environment
 - Mujoco Version Latest
 - mujoco-py version Latest

fengredrum, 2019-03-04:
Same issue occurred when using stable baseline to train Mujoco agent.
Reinstall doesn't help.
cancel progress get this message below:
`^CTraceback (most recent call last):
  File "/anaconda3/envs/py36/lib/python3.6/site-packages/lockfile/linklockfile.py", line 31, in acquire
    os.link(self.unique_name, self.lock_file)
FileExistsError: [Errno 17] File exists: '/anaconda3/envs/py36/lib/python3.6/site-packages/mujoco_py/generated/Oblivions-MBP.local-10da65c0.566-2556899123345859329' -> '/anaconda3/envs/py36/lib/python3.6/site-packages/mujoco_py/generated/mujocopy-buildlock.lock'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/anaconda3/envs/py36/lib/python3.6/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/anaconda3/envs/py36/lib/python3.6/site-packages/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mjpro_path)
  File "/anaconda3/envs/py36/lib/python3.6/site-packages/mujoco_py/builder.py", line 89, in load_cython_ext
    with LockFile(lockpath):
  File "/anaconda3/envs/py36/lib/python3.6/site-packages/lockfile/__init__.py", line 197, in __enter__
    self.acquire()
  File "/anaconda3/envs/py36/lib/python3.6/site-packages/lockfile/linklockfile.py", line 50, in acquire
    time.sleep(timeout is not None and timeout / 10 or 0.1)
KeyboardInterrupt
`

fengredrum, 2019-03-04:
Go to folder `/anaconda3/envs/py36/lib/python3.6/site-packages/mujoco_py/generated/` and delete `Oblivions-MBP.local-10da65c0.566-2556899123345859329` (in my case) that  helps. Hoping it works for you as well. @HareshMiriyala

HareshMiriyala, 2019-03-04:
@fengredrum  thanks

closed 2019-03-04 by HareshMiriyala

Wenxuan-Zhou, 2019-04-03:
I have the same issue. I tried to remove local files in the generated folder but it doesn't help.

HailinRen, 2019-04-16:
@Wenxuan-Zhou . Try to delete files end with long number strings such as "beta-System-Product-Name-56e21740.6126-3597545483291144405" and "mujocopy-buildlock.lock". At least, it works for me.

mingfeisun, 2019-04-28:
If the mujoco_py is installed in the user machine, delete the **mujocopy-buildlock.lock** and **machine-2342urwqpeur** in the following directory:

`~/.local/lib/python3.5/site-packages/mujoco_py/generated`

zuoxingdong, 2019-05-09:
@mingfeisun It works well now, but I am quite curious why these files are generated and block the Mujoco environment to be created ?

pengzhenghao, 2019-09-26:
Delete `MACHINENAME-xxxxxx` and `mujocopy-buildlock.lock` really help. I am also wondering why this happen and how to avoid it in the future.

ThibaultRouillard, 2019-11-22:
`pip3 uninstall mujoco_py && pip3 install mujoco_py`

im-Kitsch, 2021-04-28:
urn off "attach to processes automatically" could solve it although maybe not proper solution.

Check this https://youtrack.jetbrains.com/issue/PY-44481

nicolasugrinovic, 2024-01-27:
@ThibaultRouillard that worked great for me, thanks!