openai/mujoco-py #366: how to use mj_setConst

Joll123, 2019-03-12:
@everyone
How to use mj_setConst on the python side .
I want to dynamically set the position of body.
thanks

sohakes, 2019-06-12:
I found your question while searching for how to call mj_xxxx functions. I think I got mj_applyFT working based on the ray method in mjsim.pyx. I think mj_setConst is easier because it doesn't receive any parameters, is that right? In that case I think something like this in mjsim.pyx would work:

```
    def setConst(self):
        with wrap_mujoco_warning():
            mj_setConst(self.model.ptr, self.data.ptr)
```

You need to reset the mujoco-py/generated folder so cython (I think?) recompiles everything.

I think you already solved it, but it may be helpful if someone else come to this, I didn't find a guide to add new methods.

jendelel, 2019-06-28:
Hi, 

in case you haven't solve the issue. I would suggest using mujoco_py.functions.mj_setConst.

Hope this helps, 

Lukas

episodeyang, 2020-03-14:
This method is introduced in April 2019 via `self.sim.set_constant`.

Here is a quick example for how to use it:

```python
class MyEnvironment:

    def set_goal_pos(self, goal):
        self.model.body_pos[-1, 1] = goal
        self.sim.set_constants()

    def reset_model(self, x=None, goal=None):
        if goal is None:
            goal = self._get_goal()
        self.goal = goal
        if x is None:
            x = self._get_state()

        qpos = self._get_goal_state()
        self.set_goal_pos(1)
        self.set_state(qpos, self.sim.data.qvel)

        img = self.render(self.view_mode, width=self.width, height=self.height)
        if self.view_mode == "grey":
            img = img[..., None]
        self.goal_img = img.transpose(2, 0, 1) / 255

        # now reset.
        self.set_goal_pos(self.goal)
        self.sim.data.qpos[:] = x
        self.sim.data.qvel[:] = 0  # no velocity
        return self._get_obs()
```

MoritzTaylor, 2020-08-03:
@geyang Is set_constants() really needed in your case? The position of the body will change even without set_constants().
But set_constants()is there to recompute derived quantities (i.e. if you change the mass of the body the density will be recomputed if I am not wrong).
But maybe it is save to let MuJoCo recompute all values but I am not sure how this will effect performance if you have many changing values.