openai/mujoco-py #367: assert not done error in an edited humanoidstandup environment

gkcgvn, 2019-03-13:
Hi, I modified the humanoidstandup-v2, and I would like to stop the episode when some of the conditions (like stepping) are satisfied. For this reason, I placed a done condition instead of built-in done= False inside step function. When the step is taken by the humanoid, the environment is giving assert not done (from mujoco_env.py) error. When I switched only done condition to false as in v2 everything is working. Stopping the episode when the step is taken is crucial for me. How can I modify done condition without assertion error? When I remove assert not done from mujoco_py the env is not working.

gkcgvn, 2019-03-14:
Ok I solved it :)

closed 2019-03-14 by gkcgvn

tyozmen, 2020-08-27:
What caused it and how did you solve it?

TigerStone93, 2021-03-03:
@tyozmen It's because there might be some errors in 2 possible cases.
1. Check `notdone` condition in `YOUR_ENVIRONMENT_NAME.py`
2. Check `pos`, `size`, `fromto` in `YOUR_ENVIRONMENT_NAME.xml`