openai/mujoco-py #372: No matching distribution found for 'mujoco-py<2.0,>=2.0'

kracon7, 2019-03-19:
I had mujoco150 and mujoco-py 1.5 earlier
I just did `pip uninstall mujoco-py` and tried to install mujoco-py 2.0
This is the error message I got:
```
(mujoco) jc@jc-Precision-5510:~$ pip install -U 'mujoco-py<2.1,>=2.0'
Collecting mujoco-py<2.1,>=2.0
  Could not find a version that satisfies the requirement mujoco-py<2.1,>=2.0 (from versions: 0.2.4, 0.2.5, 0.3.0, 0.3.1, 0.3.2, 0.4.0, 0.4.1, 0.4.2, 0.4.3, 0.4.4, 0.5.0, 0.5.1, 0.5.2, 0.5.3, 0.5.4, 
0.5.5, 0.5.6, 0.5.7, 1.5.0.0.1, 1.50.0.1, 1.50.0.2, 1.50.0.3, 1.50.0.4, 1.50.0.5, 1.50.0.6, 1.50.0.7, 1.50.0.8, 1.50.0.9, 1.50.0.10, 1.50.0.11, 1.50.0.12, 1.50.0.13, 1.50.0.14, 1.50.1.0, 1.50.1.1, 1.50.1.2, 1.50.1.3, 1.50.1.4, 1.50.1.5, 1.50.1.6, 1.50.1.7, 1.50.1.8, 1.50.1.9, 1.50.1.10, 1.50.1.11, 1.50.1.12, 1.50.1.13, 1.50.1.14, 1.50.1.15, 1.50.1.16, 1.50.1.17, 1.50.1.18, 1.50.1.19, 1.50.1.20,
 1.50.1.21, 1.50.1.22, 1.50.1.23, 1.50.1.24, 1.50.1.25, 1.50.1.26, 1.50.1.27, 1.50.1.28, 1.50.1.29, 1.50.1.30, 1.50.1.31, 1.50.1.32, 1.50.1.33, 1.50.1.34, 1.50.1.35, 1.50.1.36, 1.50.1.37, 1.50.1.38, 1.50.1.39, 1.50.1.40, 1.50.1.41, 1.50.1.42, 1.50.1.43, 1.50.1.44, 1.50.1.45, 1.50.1.46, 1.50.1.47,
 1.50.1.48, 1.50.1.49, 1.50.1.50, 1.50.1.51, 1.50.1.52, 1.50.1.53, 1.50.1.54, 1.50.1.55, 1.50.1.56, 1.50.1.59, 1.50.1.62, 1.50.1.63, 1.50.1.64, 1.50.1.65a1, 1.50.1.65, 1.50.1.65.1, 1.50.1.65.2, 1.50.1.65.3, 1.50.1.66, 1.50.1.67, 1.50.1.68, 2.0.2.0a1)
No matching distribution found for mujoco-py<2.1,>=2.0
```

 - OS: Ubuntu 16.04
 - Python: 3.5.2
 - Mujoco Version: mujoco200
 - mujoco-py version 2.0

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
/home/jc/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/home/jc/.mujoco/mjpro150/bin:/usr/lib/nvidia-384

 - output of: `echo $HOME`
/home/jc

 - output of: `echo $USER`
jc

hyc6668378, 2019-03-19:
I have the same problem.
error message I got:

` No matching distribution found for mujoco-py<2.1,>=2.0
`

- OS: Ubuntu 16.04
- Python: 3.6.7
- Mujoco Version: mujoco200
### Environment

- output of: `echo $LD_LIBRARY_PATH`
/home/hsk/.mujoco/mjpro150/bin

- output of:`echo $HOME`
/home/hsk

hyc6668378, 2019-03-19:
It's looks like that `2.0.2.0a1` is avaliable.

cyrilzakka, 2019-03-19:
I'm having a different problem. Despite updating my setup.py to the following: 

```
install_requires=['gym>=0.12.0', 
                  'mujoco-py>=2.0.2.0a1', 
      		  'imageio']
```

it seems that gym still uses mjpro150 instead of mujoco200 despite downloading the newest version and replacing the old one. This results in the following error:
```
Import error. Trying to rebuild mujoco_py.
running build_ext
building 'mujoco_py.cymj' extension
/usr/local/bin/gcc-6 -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/cyril/miniconda3/include -arch x86_64 -I/Users/cyril/miniconda3/include -arch x86_64 -DONMAC -Iminiconda3/lib/python3.7/site-packages/mujoco_py-1.50.1.59-py3.7.egg/mujoco_py -I/Users/cyril/miniconda3/lib/python3.7/site-packages/mujoco_py-1.50.1.59-py3.7.egg/mujoco_py -I/Users/cyril/.mujoco/mjpro150/include -I/Users/cyril/miniconda3/lib/python3.7/site-packages/numpy/core/include -I/Users/cyril/miniconda3/include/python3.7m -c /Users/cyril/miniconda3/lib/python3.7/site-packages/mujoco_py-1.50.1.59-py3.7.egg/mujoco_py/cymj.c -o /Users/cyril/miniconda3/lib/python3.7/site-packages/mujoco_py-1.50.1.59-py3.7.egg/mujoco_py/generated/_pyxbld_1.50.1.59_37_macextensionbuilder/temp.macosx-10.7-x86_64-3.7/Users/cyril/miniconda3/lib/python3.7/site-packages/mujoco_py-1.50.1.59-py3.7.egg/mujoco_py/cymj.o -fopenmp -w
/Users/cyril/miniconda3/lib/python3.7/site-packages/mujoco_py-1.50.1.59-py3.7.egg/mujoco_py/cymj.c:620:21: fatal error: mjmodel.h: No such file or directory
 #include "mjmodel.h"

```

Krasaa, 2019-03-21:
@hyc6668378 
Hi, I have exactly the same environment as yours.
However, I tried `pip install -U 'mujoco-py 2.0.2.0a1'` which doesn't help. How do you solve this problem?

hyc6668378, 2019-03-22:
@Krasaa have you try` pip install -U 'mujoco-py<2.1,>=2.0.2.0a1'`?

MillionIntegrals, 2019-03-22:
Previously released version was an alpha named '2.0.2.0a1'. I've just released '2.0.2.0' so that installing `mujoco-py<2.1,>=2.0` should work now. I think the gym may still depend on the previous mujoco version, I'll take a look there.

MillionIntegrals, 2019-03-23:
Works for me with gym with no problems, the version requirement there is `>=1.50`.

closed 2019-03-25 by MillionIntegrals