openai/mujoco-py #373: Can't render the model which include deformable objects

hyc6668378, 2019-03-19:
**Describe the bug**
The `.xml` file which I used include cloth (`composite`).
If I render the model,like 

```
viewer = MjViewer(sim)
viewer.render()
```
It will come out the bug below

**Expected behavior**
I really want to use the Deformable objects in simulation.
and I want to observe what happened by render 

**Error Messages**
```
Creating window glfw
Segmentation fault (core dumped)
```
**Desktop:**
 - OS:  ubuntu 16.04
 - Python Version:  3.6.7
 - Mujoco Version : 200
 - mujoco-py version : 2.0.2.0a1

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
/home/hsk/.mujoco/mujoco200/bin

 - output of: `echo $HOME`
/home/hsk

 - output of: `echo $USER`
hsk

evelyndelia, 2019-04-09:
I am having the exact same issue. With the deformable object, I get a segmentation fault. With the same .xml and the deformable object commented out, I do not get the error. 

I have the same Desktop profile as above.

Has this bug been identified yet?

fhln, 2019-04-10:
this same error: Segmentation fault. Please help.

wutter, 2019-05-17:
Same problem here. Both in virtual Ubuntu 16.04, python3.7, mujoco2 and also in macos 10.14.4, python3.7, mujoco2. When checking GDB in ubuntu I got:
```
(gdb) run test.py 
Starting program: /usr/bin/python3.7 test.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 3225]
[Detaching after fork from child process 3226]
[Detaching after fork from child process 3227]
[Detaching after fork from child process 3228]
[Detaching after fork from child process 3234]
[Detaching after fork from child process 3240]
[Detaching after fork from child process 3246]
Creating window glfw

Program received signal SIGSEGV, Segmentation fault.
__memset_avx2 () at ../sysdeps/x86_64/multiarch/memset-avx2.S:135
135	../sysdeps/x86_64/multiarch/memset-avx2.S: Directory or file does not exist.
```

From /proc/cpuinfo:
```
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 70
model name	: Intel(R) Core(TM) i7-4870HQ CPU @ 2.50GHz
stepping	: 1
cpu MHz		: 2494.226
cache size	: 6144 KB
physical id	: 0
siblings	: 1
core id		: 0
cpu cores	: 1
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 syscall nx rdtscp lm constant_tsc rep_good nopl xtopology nonstop_tsc cpuid tsc_known_freq pni pclmulqdq monitor ssse3 cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx rdrand hypervisor lahf_lm abm invpcid_single pti fsgsbase avx2 invpcid flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds
bogomips	: 4988.45
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

```

wutter, 2019-05-20:
Ok, found where the problem is, made fork:
https://github.com/wutter/mujoco-py/blob/master/mujoco_py/mjrendercontext.pyx
`pip3 install --user --force-reinstall --no-compile git+https://github.com/wutter/mujoco-py/tree/5ba1983931925b27e9b2e542c1a709b3ac106891`

if you try file from previous post, you can see it falls on update:
```
Creating window glfw
Init render
Will set buffer size
Will update scene
Segmentation fault: 11
```
With that line commented out, it gets to render and fall there (as there is nothing to render I guess).

Next might be checking the differences between vars for soft-body scene vs non-soft body one and vars from basic.cpp from mujoco installation for the soft-body model (because it works there), but I do not know how to do that.

wutter, 2019-05-22:
Until solved, going with tempomary solution: visual/viewer in C++ connected via UDP to python. Qpos are than being send as timeframes and viewer loads them.

wutter, 2019-05-29:
Found out where is the problem, its not deformable object itself, but the skin. It seems like skins are not supported for some reason.

yilin-wu98, 2019-07-15:
I encounter the same bug and after removing the skin part, the render works, but I need the skin attribute to make the structure look like a cloth. I am just wondering if anyone knows in which way I can use the skin because without the skin, I can only see the internal structure of a cloth composite object, lacking the visual attribute.

robotszu, 2019-08-07:
I also have the same problem. It works without skin. But I need skin. How to add skin in mujoco-py? Really appreciate it if anyone helps.

usut, 2019-08-07:
I have the same problem. It works without skin. But I need skin. How to add skin in mujoco-py? Really appreciate it if anyone helps.

usut, 2019-08-08:
> Found out where the problem is. Its not deformable object itself, but the skin. It seems like skins are not supported for some reason. If you remove skin, even the carpet works.

Hello, have you solved this problem?

wutter, 2019-08-09:
I did not bother to solve in python, working solution if you need the viewer is https://github.com/openai/mujoco-py/issues/373#issuecomment-494696192 I can post the code public (will update this comment with git link)

robotszu, 2019-08-09:
> I did not bother to solve in python, working solution if you need the viewer is [#373 (comment)](https://github.com/openai/mujoco-py/issues/373#issuecomment-494696192) I can post the code public (will update this comment with git link)

It would be helpful if you could post the code with a link. Thanks :)

usut, 2019-08-09:
> I did not bother to solve in python, working solution if you need the viewer is [#373 (comment)](https://github.com/openai/mujoco-py/issues/373#issuecomment-494696192) I can post the code public (will update this comment with git link)

It would be helpful if you could post the code with a link. Thanks :)

usut, 2019-08-09:
I got it. Thank you very much.

coffeiersama, 2019-08-10:
> I did not bother to solve in python, working solution if you need the viewer is [#373 (comment)](https://github.com/openai/mujoco-py/issues/373#issuecomment-494696192) I can post the code public (https://github.com/wutter/mujoco-py-to-c-viewer)

hi！ 
Thanks for your codes!
I replace basic.cpp in my mujoco200 with your viewer.cpp and then make the file, 
then I try to run my project with skin, but there is still the same error,
Is there any problem in my method of using your codes??

wutter, 2019-08-10:
Sorry, I did not put much docs on how to use.

1. Modify your model (copy), so it does not use skin.
2. Use your python code with modle without skin to generate the sim.data.qpos (and than export it to csv by file.writerow(sim.data.qpos).
3. Use the csv with qpos + model with skin as inputs for the viewer (.cpp). Now you should be able to see the output of the simulation even with the skin.

There is also space for improvement, so it does export active forces, so you could display them, but I did not bother as it was not importatnt for my use.

coffeiersama, 2019-08-15:
> Sorry, I did not put much docs on how to use.
> 
>     1. Modify your model (copy), so it does not use skin.
> 
>     2. Use your python code with the model without skin to generate the sim.data.qpos (and than export it to csv by file.writerow(sim.data.qpos).
> 
>     3. Use the csv with qpos + model with skin as inputs for the viewer (.cpp).
> 
> 
> There is also space for improvement, it does export active forces, so you could display them, but I did not bother as it was not importatnt for my use.

Thanks for the quick reply,
I have followed the instruction and get to see the cloth skin in a single frame by running the viewer,
I was  wondering whether i can use this to show a continuous animation by updating the csv file?

wutter, 2019-08-15:
Actually, it is already continuous in mujoco sense, that it generates 1 frame each timestep (setted up in mujoco)

wutter, 2019-08-17:
[443#issuecomment-521965723](https://github.com/openai/mujoco-py/issues/443#issuecomment-521965723) (just so its all at one place).
Ah I see, like what do you mean continuous. That I did not solve. Most portable solution would probably be to use socket, but easier solution might be to have a lock file (random file) and the output, so the C-viewer would pre-load generated data when non-locked and the python would just generate some frame blocks (assuming the generator is faster than realtime) and while doing so, create the lock file.

coffeiersama, 2019-08-17:
> [443#issuecomment-521965723](https://github.com/openai/mujoco-py/issues/443#issuecomment-521965723) (just so its all at one place).
> Ah I see, like what do you mean continuous. That I did not solve. [...]

Adam, thanks for your prompt response. 

What I want to achieve is to detect the corner of a cloth when the cloth is being manipulated. We need the skin displayed all the time. It is infeasible to pause the motion and display skin as we do not know when to pause. Here, the cloth is being maniplated and move continuously.

Do you mean that this kind of continous motion has not been fixed?

brinij, 2019-12-03:
Is anyone working on solving this problem?

fantasyRqg, 2020-04-15:
bug report
http://www.mujoco.org/forum/index.php?threads/crash-mac-render-softcylinder-xml.4159/

fantasyRqg, 2020-04-16:
after this change. success to use model contain skin tag

jangirrishabh, 2020-04-20:
Looking for solutions to this problem, the above mentioned solution does not work for me @fantasyRqg

fantasyRqg, 2020-04-21:
Install insturction

```
pip3 uninstall  -y mujoco_py

git clone git@github.com:fantasyRqg/mujoco-py.git

cd mujoco-py

python3 setup.py build

python3 setup.py install
```

**make sure you have installed gcc**
Mac OS :  `brew install gcc`
Ubuntu: `sudo apt-get install gcc`

[image]

TMmichi, 2020-07-28:
> Install instructions
> 
> ```
> pip3 uninstall  -y mujoco_py
> 
> git clone git@github.com:fantasyRqg/mujoco-py.git
> 
> cd mujoco-py
> 
> python3 setup.py build
> 
> python3 setup.py install
> ```
> 
> **make sure you have installed the gcc**
> Mac OS : `brew install gcc`
> Ubuntu: `sudo apt-get install gcc`
> 
> Here is softbox demo of mujoco
> [image]
> 
> @jangirrishabh

Many thanks. You are a genius