openai/mujoco-py #374: Install mujoco with error "/usr/bin/ld: cannot find -lGL" and "gcc" error

JuliusVsi, 2019-03-24:
**Describe the bug**
Install MuJoCo via pip3 install -U 'pip3 install -U 'mujoco-py<2.1,>=2.0', I got
    /usr/bin/ld: cannot find -lGL
    collect2: error: ld returned 1 exit status
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
I could open the simulation by running ../mjpro150/bin: $ ./simulate ../model/humanoid.xml

**To Reproduce**
1. Download the MuJoCo version 2.0 binaries for Linux or OSX.
2. Unzip the downloaded mujoco200 directory into ~/.mujoco/mujoco200, and place your license key (the mjkey.txt file from your email) at ~/.mujoco/mjkey.txt.
3. export LD_LIBRARY_PATH=~/.mujoco/mujoco200/bin${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
export MUJOCO_KEY_PATH=~/.mujoco${MUJOCO_KEY_PATH}
4. pip3 install -U 'mujoco-py<2.1,>=2.0

**Error Messages**
    /usr/bin/ld: cannot find -lGL
    collect2: error: ld returned 1 exit status
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-g1xlgknn/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-gv_q8yd9-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-g1xlgknn/mujoco-py/

**Desktop (please complete the following information):**
 - Ubuntu 18.04
 - Python 3.6.8
 - Mujoco Version 2.0
 - mujoco-py version [2.0]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`$LD_LIBRARY_PATH ~/.mujoco/mujoco200/bin
 - output of: `echo $HOME`/home/julius
 - output of: `echo $USER /julius
[image]

MillionIntegrals, 2019-03-25:
This seems to be an issue with ubuntu packages failing to create a proper link to libGL.so, although I don't know much about the underlying reasons why it's so.

It's been discussed already in a few places:
https://github.com/openai/mujoco-py/issues/341
https://stackoverflow.com/questions/18406369/qt-cant-find-lgl-error/54397235#54397235
https://stackoverflow.com/questions/33447653/usr-bin-ld-cannot-find-lgl-ubuntu-14-04
http://techtidings.blogspot.com/2012/01/problem-with-libglso-on-64-bit-ubuntu.html

I'll try to put some information in the Readme about this problem

Usually this solves the issue:
```
apt install libgl1-mesa-glx --reinstall
```

If not, then it may need to be followed by creating the link directly:

```
ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1 /usr/lib/x86_64-linux-gnu/libGL.so
```

closed 2019-04-02 by MillionIntegrals

BeBraveBeCurious, 2022-05-17:
Is there any recommended solution without root or sudo? Thank you.

lafmdp, 2022-07-20:
> This seems to be an issue with ubuntu packages failing to create a proper link to libGL.so, although I don't know much about the underlying reasons why it's so.
>
> It's been discussed already in a few [...]

This works for me!