openai/mujoco-py #378: OpenGL version 1.5 or higher required

cbjtu, 2019-03-28:
I installed the mujoco and mujoco_py normally, and I can run the example at the README.

But at `~/.mujoco/mujoco200/bin` I try to run `./simulate ../model/humanoid.xml`, the server said that
```
MuJoCo Pro version 2.00
ERROR: OpenGL version 1.5 or higher required

Press Enter to exit ...
```
And get a Black Screen with nothing.

But I have installed the OpenGL related library, and there's rare issue about that. What can I do?

Baichenjia, 2019-04-02:
I meet the same problem when I use mujco150, the error is 
```
>>> import gym
>>> env = gym.make("FetchPush-v1")
>>> obs = env.reset()
>>> env.render()
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
Creating window glfw
ERROR: OpenGL version 1.5 or higher required
```
How can I solve it?

Baichenjia, 2019-04-02:
I use Ubuntu 16.04, Anaconda python 3.6.1, mujoco-py 1.50.1.56  , gym 0.10.5 ,
Nvidia driver 410.73 and CUDA 10.  
I can `import mujoco_py` and run `env.step(action)` successfully, but I cannot run `env.render`.

cbjtu, 2019-04-04:
I can train the model, but can't visualize it

yyf17, 2019-05-09:
I meet the same problem when I use mujco150,
>>> import gym
>>> env = gym.make("FetchPush-v1")
>>> obs = env.reset()
>>> env.render()

 the error is
GLFW error (code %d): %s 65544 b'X11: RandR gamma ramp support seems broken'
GLFW error (code %d): %s 65542 b'EGL: Failed to get EGL display: Success'
GLFW error (code %d): %s 65544 b'X11: RandR monitor support seems broken'
Creating window glfw
ERROR: OpenGL version 1.5 or higher required

Press Enter to exit ...

Rowing0914, 2019-12-03:
any thoughts?

apetrasc, 2023-05-29:
try "sudo gedit /etc/environment" and add one sentense below it
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/{username}/.mujoco/mujoco210/bin:/usr/lib/nvidia
(you have to replace username to your user name)
after that, try these commands
sudo apt-get install libglew-dev
sudo apt install libopengl0
and then, I can use this
./simulate ../model/humanoid.xml 
command. and got this 
[image]