openai/mujoco-py #379: dlopen: cannot load any more object with static TL

liziniu, 2019-03-28:
**Describe the bug**
I install mujoco_py=2.0, gym=0.10.11, python=3.6.5. Then, I use gym.make() to create mujoco tasks like "ant-v2", "halfcheetah-v2", and etc. Everything is ok when I use gym alone. However, when I run codes from baselines. It turns out that I got an import error from mujoco_py, saying that there is an error from rebuilding mujoco_py. The error is:

`dlopen: cannot load any more object with static TLS`

> Import error. Trying to rebuild mujoco_py.
running build_ext
building 'mujoco_py.cymj' extension
gcc -pthread -B /home/lizn/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py -I/home/lizn/.mujoco/mujoco200/include -I/home/lizn/anaconda3/lib/python3.6/site-packages/numpy/core/include -I/home/lizn/anaconda3/include/python3.6m -c /home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/cymj.c -o /home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/cymj.o -fopenmp -w
gcc -pthread -B /home/lizn/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py -I/home/lizn/.mujoco/mujoco200/include -I/home/lizn/anaconda3/lib/python3.6/site-packages/numpy/core/include -I/home/lizn/anaconda3/include/python3.6m -c /home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/gl/osmesashim.c -o /home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/gl/osmesashim.o -fopenmp -w
gcc -pthread -shared -B /home/lizn/anaconda3/compiler_compat -L/home/lizn/anaconda3/lib -Wl,-rpath=/home/lizn/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/cymj.o /home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxcpuextensionbuilder/temp.linux-x86_64-3.6/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/gl/osmesashim.o -L/home/lizn/.mujoco/mujoco200/bin -Wl,-R/home/lizn/.mujoco/mujoco200/bin -lmujoco200 -lglewosmesa -lOSMesa -lGL -o /home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxcpuextensionbuilder/lib.linux-x86_64-3.6/mujoco_py/cymj.cpython-36m-x86_64-linux-gnu.so -fopenmp
Traceback (most recent call last):
  File "/home/lizn/package/mujoco/gym-0.10.11/gym/envs/mujoco/mujoco_env.py", line 11, in <module>
    import mujoco_py
  File "/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/builder.py", line 503, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/builder.py", line 106, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/home/lizn/anaconda3/lib/python3.6/site-packages/mujoco_py/builder.py", line 124, in load_dynamic_ext
    return loader.load_module()
ImportError: dlopen: cannot load any more object with static TLS 

**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Error Messages**
Including more/longer error messages gives us more information to work with.

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 14]
 - Python Version [e.g. 3.6.5]
 - Mujoco Version [e.g. 200]
 - mujoco-py version [e.g. 0.10.11]

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
/usr/local/cuda-8.0/lib64:/usr/local/cuda-8.0/lib64:$LD_LIBRARY_PATH:/home/lizn/.mujoco/mjpro150/bin:/home/lizn/.mujoco/mjpro150/bin:/home/lizn/.mujoco/mujoco200/bin:/home/lizn/.mujoco/mjpro150/bin:/home/lizn/.mujoco/mujoco200/bin
 - output of: `echo $HOME`
 - output of: `echo $USER`

**Additional context**
Add any other context about the problem here.

liziniu, 2019-03-31:
I solve this by installing tensorflow-gpu rather than tensorflow-cpu.

closed 2019-04-02 by MillionIntegrals

LxzGordon, 2021-11-25:
same problem here. Seems that importing mujoco_py from another package will lead to some bug during importing cymj