openai/mujoco-py #380: Black screen

richardrl, 2019-03-31:
**Describe the bug**
Screen is black for any mujoco-py gym environments
**To Reproduce**
Ubuntu 18.04
Nvidia 1080 Ti
Mujoco 1.5 or 2.0

Simulation runs (prints outputs), but screen is completely black
**Expected behavior**
Should show simulation

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
:/home/richard/.mujoco/mjpro150/bin:/usr/local/cuda/lib64:/usr/local/cuda/extras/CUPTI/lib64:/home/richard/.mujoco/mujoco200/bin

 - output of: `echo $HOME`
/home/richard
 - output of: `echo $USER`
richard

richardrl, 2019-03-31:
If I set LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/x86_64-linux-gnu/libGL.so

Instead of a black screen, I get the:
ERROR: GLEW initalization error: Missing GL version

Press Enter to exit ...

tkelestemur, 2019-03-31:
Same here. I get a black screen and the same error.

My setup: 
Ubuntu 18.04, RTX2080Ti. MuJoCo 200.
mujoco-py version 2.0.2.0
gym version 0.12.1

underactuated, 2019-04-25:
I had the same problem (ubuntu 18.04). I am no expert, but reading related discussions helped me solve it. Solution:

1) install GLEW: 
sudo apt-get install libglew-dev

2) add to .bashrc:
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so

richardrl, 2019-04-26:
@underactuated As stated earlier in the thread these are not solutions for me

tkelestemur, 2019-07-07:
@underactuated's solution works for me but I'm wondering if there is a better solution. (I don't wanna put LD_PRELOAD in bashrc.)

mh-mu, 2022-09-20:
Has anyone been able to resolve this issue? I have tried all methods in this thread but none worked for me so far.