openai/mujoco-py #381: ERROR: Shadow framebuffer is not complete, error 0x8cd7

siekmanj, 2019-04-04:
**Describe the bug**
When using MjSim.render(), the interpreter crashes.

**To Reproduce**
`python3`
`>>> import mujoco_py as mj`
`>>> model = mj.load_model_from_path("./humanoid.xml")`
`>>> sim = mj.MjSim(model)`
`>>> sim.render()`
`ERROR: Shadow framebuffer is not complete, error 0x8cd7`
``
`Press Enter to exit ...`

**Error Messages**
This is the only message I got:

`ERROR: Shadow framebuffer is not complete, error 0x8cd7`

**Desktop (please complete the following information):**
 - OS: Linux 4.14.109-1-MANJARO, 64-bit
 - Python Version: 3.7.3
 - Mujoco Version 2.0
 - mujoco-py version: 2.0.2.0

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
`/home/jonah/.mujoco/mjpro150/bin/:/home/jonah/.mujoco/mujoco200/bin`

 - output of: `echo $HOME`
`/home/jonah`

 - output of: `echo $USER`
`jonah`

Any help would be much appreciated.

siekmanj, 2019-04-05:
I was able to work around this issue by using a different distro (Ubuntu 18.04). I found a lot of evidence that my graphics card (Radeon R9 380) and linux distro (Manjaro 4.1x) do not play nicely with another. I suspect rendering via the CPU would have fixed my issue as well, but I wasn't able to find instructions on how to do that.

closed 2019-04-05 by siekmanj

mrahtz, 2021-11-09:
I had this issue running mujoco-py in CPU-only Docker. I was similarly able to fix it switching from a Docker image based on `miniconda3` to one based on `ubuntu:20.04`.