openai/mujoco-py #382: universe installation failed due to mujoco

weihangChen, 2019-04-07:
**Describe the bug**
the problem is same as 
https://github.com/openai/mujoco-py/issues/317

mujoco200 install works, but the docker image still complains that 
I have Ubuntu 18.04 and managed to install mujoco200 `pip install -U mujoco-py`, I did not receive license yet, so I cant test mujoco yet. But the license should not be a problem for installing univers, so I run `ocker build -t universe .` please go to Error Messages section to view the detail.

**Error Messages**
```
Sending build context to Docker daemon  2.218MB
Step 1/4 : FROM ubuntu:16.04
 ---> 9361ce633ff1
Step 2/4 : RUN apt-get update     && apt-get install -y libav-tools     python3-numpy     python3-scipy     python3-setuptools     python3-pip     libpq-dev     libjpeg-dev     curl     cmake     swig     python3-opengl     libboost-all-dev     libsdl2-dev     wget     unzip     git     golang     net-tools     iptables     libvncserver-dev     software-properties-common     && apt-get clean     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 83b9fff605df
Step 3/4 : RUN ln -sf /usr/bin/pip3 /usr/local/bin/pip     && ln -sf /usr/bin/python3 /usr/local/bin/python     && pip install -U pip
 ---> Using cache
 ---> edc5b2f5ebd7
Step 4/4 : RUN pip install gym[all]
 ---> Running in fc6591f05de1
Collecting gym[all]
  Downloading https://files.pythonhosted.org/packages/7b/57/e2fc4123ff2b4e3d61ae9b3d08c6878aecf2d5ec69b585ed53bc2400607f/gym-0.12.1.tar.gz (1.5MB)
Requirement already satisfied: scipy in /usr/lib/python3/dist-packages (from gym[all]) (0.17.0)
Requirement already satisfied: numpy>=1.10.4 in /usr/lib/python3/dist-packages (from gym[all]) (1.11.0)
Collecting requests>=2.0 (from gym[all])
  Downloading https://files.pythonhosted.org/packages/7d/e3/20f3d364d6c8e5d2353c72a67778eb189176f08e873c9900e10c0287b84b/requests-2.21.0-py2.py3-none-any.whl (57kB)
Collecting six (from gym[all])
  Downloading https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting pyglet>=1.2.0 (from gym[all])
  Downloading https://files.pythonhosted.org/packages/1c/fc/dad5eaaab68f0c21e2f906a94ddb98175662cc5a654eee404d59554ce0fa/pyglet-1.3.2-py2.py3-none-any.whl (1.0MB)
Requirement already satisfied: PyOpenGL in /usr/lib/python3/dist-packages (from gym[all]) (3.0.2)
Collecting mujoco_py<2.1,>=1.50 (from gym[all])
  Downloading https://files.pythonhosted.org/packages/9e/ea/3fa08305b80346a787bae7b45eaa8e11d4187f85e012b95ad7bcd97f8336/mujoco-py-2.0.2.0.tar.gz (759kB)
Collecting imageio (from gym[all])
  Downloading https://files.pythonhosted.org/packages/af/0a/943c965d372dae0b1f1482677d29030ab834351a61a9a632fd62f27f1523/imageio-2.5.0-py3-none-any.whl (3.3MB)
Collecting box2d-py>=2.3.5 (from gym[all])
  Downloading https://files.pythonhosted.org/packages/c4/59/58faf9c8277efd22509db7fc068d971d894ebf650ff73770401a4d5bed78/box2d_py-2.3.8-cp35-cp35m-manylinux1_x86_64.whl (448kB)
Collecting atari_py>=0.1.4 (from gym[all])
  Downloading https://files.pythonhosted.org/packages/90/eb/1957fab30160edd96da368874dbfccbf5cae2b7bc07f72772a378eaff351/atari_py-0.1.7-cp35-cp35m-manylinux1_x86_64.whl (2.6MB)
Collecting Pillow (from gym[all])
  Downloading https://files.pythonhosted.org/packages/de/62/6358ccd27b9e340affaa29e3e678b5d93a4b17f6bd7533fd6e6e87930b12/Pillow-6.0.0-cp35-cp35m-manylinux1_x86_64.whl (2.0MB)
Collecting certifi>=2017.4.17 (from requests>=2.0->gym[all])
  Downloading https://files.pythonhosted.org/packages/60/75/f692a584e85b7eaba0e03827b3d51f45f571c2e793dd731e598828d380aa/certifi-2019.3.9-py2.py3-none-any.whl (158kB)
Collecting idna<2.9,>=2.5 (from requests>=2.0->gym[all])
  Downloading https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl (58kB)
Collecting chardet<3.1.0,>=3.0.2 (from requests>=2.0->gym[all])
  Downloading https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl (133kB)
Collecting urllib3<1.25,>=1.21.1 (from requests>=2.0->gym[all])
  Downloading https://files.pythonhosted.org/packages/62/00/ee1d7de624db8ba7090d1226aebefab96a2c71cd5cfa7629d6ad3f61b79e/urllib3-1.24.1-py2.py3-none-any.whl (118kB)
Collecting future (from pyglet>=1.2.0->gym[all])
  Downloading https://files.pythonhosted.org/packages/90/52/e20466b85000a181e1e144fd8305caf2cf475e2f9674e797b222f8105f5f/future-0.17.1.tar.gz (829kB)
Collecting glfw>=1.4.0 (from mujoco_py<2.1,>=1.50->gym[all])
  Downloading https://files.pythonhosted.org/packages/5d/65/c6275744a01425195f1f446e022e5dfa6497aa68479a3952e434e04b2fa0/glfw-1.7.1.tar.gz
Collecting Cython>=0.27.2 (from mujoco_py<2.1,>=1.50->gym[all])
  Downloading https://files.pythonhosted.org/packages/c2/34/99ced126b3f41a908d8883570a67fbf900f10eea3cfdd11e388eb8ae9aac/Cython-0.29.6-cp35-cp35m-manylinux1_x86_64.whl (2.0MB)
Collecting cffi>=1.10 (from mujoco_py<2.1,>=1.50->gym[all])
  Downloading https://files.pythonhosted.org/packages/5b/44/fdae2a8f66af426055f9b6fff0b155217081eddaf08a3df79ca11fe05bda/cffi-1.12.2-cp35-cp35m-manylinux1_x86_64.whl (428kB)
Collecting lockfile>=0.12.2 (from mujoco_py<2.1,>=1.50->gym[all])
  Downloading https://files.pythonhosted.org/packages/c8/22/9460e311f340cb62d26a38c419b1381b8593b0bb6b5d1f056938b086d362/lockfile-0.12.2-py2.py3-none-any.whl
Collecting pycparser (from cffi>=1.10->mujoco_py<2.1,>=1.50->gym[all])
  Downloading https://files.pythonhosted.org/packages/68/9e/49196946aee219aead1290e00d1e7fdeab8567783e83e1b9ab5585e6206a/pycparser-2.19.tar.gz (158kB)
Building wheels for collected packages: gym, mujoco-py, future, glfw, pycparser
  Building wheel for gym (setup.py): started
  Building wheel for gym (setup.py): finished with status 'done'
  Stored in directory: /root/.cache/pip/wheels/57/b0/13/4153e1acab826fbe612c95b1336a63a3fa6416902a8d74a1b7
  Building wheel for mujoco-py (setup.py): started
  Building wheel for mujoco-py (setup.py): finished with status 'error'
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-2yzbct5r/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-58sck1ff --python-tag cp35:
  running bdist_wheel
  running build
[... 96 lines cut ...]
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-2yzbct5r/mujoco-py/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-v_9pw6yx/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-2yzbct5r/mujoco-py/
The command '/bin/sh -c pip install gym[all]' returned a non-zero code: 1

```

**Desktop (please complete the following information):**
 - OS: ubuntu18.04
 - Python Version Python 3.6.5 :: Anaconda, Inc.
 - Mujoco Version 200
 - mujoco-py version 2.0.2.0

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`
:/home/marc/.mujoco/mujoco200/bin

 - output of: `echo $HOME`
/home/marc

 - output of: `echo $USER`
marc

**Additional context**
Add any other context about the problem here.

hougiebear, 2019-06-07:
Hi I am having the same issue. Trying to find a workaround but no joy yet

Shashank001122, 2019-10-02:
Hi Everyone! Did you find any workaround, I am also stuck.

weihangChen, 2019-10-03:
i think the project is kind of dead, try to find some other environment to work with.