openai/mujoco-py #383: GPU compilation broken with cuda 10: conflicting types for ‘PFNGLFRAGMENTLIGHTFVSGIXPROC’

hartikainen, 2019-04-08:
**Describe the bug**
I'm trying to install and run mujoco-py with GPU. The compilation fails with the following error:
```
Compiling /mujoco_py/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /mujoco_py/mujoco_py/cymj.pyx
running build_ext
building 'mujoco_py.cymj' extension
creating /mujoco_py/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxgpuextensionbuilder
creating /mujoco_py/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6
creating /mujoco_py/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/mujoco_py
creating /mujoco_py/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/mujoco_py/mujoco_py
creating /mujoco_py/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/mujoco_py/mujoco_py/gl
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/mujoco_py/mujoco_py -I/root/.mujoco/mujoco200/include -I/env/lib/python3.6/site-packages/numpy/core/include -I/mujoco_py/mujoco_py/vendor/egl -I/usr/include/python3.6m -I/env/include/python3.6m -c /mujoco_py/mujoco_py/cymj.c -o /mujoco_py/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/mujoco_py/mujoco_py/cymj.o -fopenmp -w
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Imujoco_py -I/mujoco_py/mujoco_py -I/root/.mujoco/mujoco200/include -I/env/lib/python3.6/site-packages/numpy/core/include -I/mujoco_py/mujoco_py/vendor/egl -I/usr/include/python3.6m -I/env/include/python3.6m -c /mujoco_py/mujoco_py/gl/eglshim.c -o /mujoco_py/mujoco_py/generated/_pyxbld_2.0.2.0_36_linuxgpuextensionbuilder/temp.linux-x86_64-3.6/mujoco_py/mujoco_py/gl/eglshim.o -fopenmp -w
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:7:0:
/usr/include/GL/glext.h:12066:25: error: conflicting types for ‘PFNGLFRAGMENTLIGHTFVSGIXPROC’
 typedef void (APIENTRYP PFNGLFRAGMENTLIGHTFVSGIXPROC) (GLenum light, GLenum pname, const GLfloat *params);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:4:0:
/usr/include/GL/glew.h:16092:28: note: previous declaration of ‘PFNGLFRAGMENTLIGHTFVSGIXPROC’ was here
 typedef void (GLAPIENTRY * PFNGLFRAGMENTLIGHTFVSGIXPROC) (GLenum light, GLenum pname, GLfloat* params);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:7:0:
/usr/include/GL/glext.h:12068:25: error: conflicting types for ‘PFNGLFRAGMENTLIGHTIVSGIXPROC’
 typedef void (APIENTRYP PFNGLFRAGMENTLIGHTIVSGIXPROC) (GLenum light, GLenum pname, const GLint *params);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:4:0:
/usr/include/GL/glew.h:16094:28: note: previous declaration of ‘PFNGLFRAGMENTLIGHTIVSGIXPROC’ was here
 typedef void (GLAPIENTRY * PFNGLFRAGMENTLIGHTIVSGIXPROC) (GLenum light, GLenum pname, GLint* params);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:7:0:
/usr/include/GL/glext.h:12070:25: error: conflicting types for ‘PFNGLFRAGMENTLIGHTMODELFVSGIXPROC’
 typedef void (APIENTRYP PFNGLFRAGMENTLIGHTMODELFVSGIXPROC) (GLenum pname, const GLfloat *params);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:4:0:
/usr/include/GL/glew.h:16088:28: note: previous declaration of ‘PFNGLFRAGMENTLIGHTMODELFVSGIXPROC’ was here
 typedef void (GLAPIENTRY * PFNGLFRAGMENTLIGHTMODELFVSGIXPROC) (GLenum pname, GLfloat* params);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:7:0:
/usr/include/GL/glext.h:12072:25: error: conflicting types for ‘PFNGLFRAGMENTLIGHTMODELIVSGIXPROC’
 typedef void (APIENTRYP PFNGLFRAGMENTLIGHTMODELIVSGIXPROC) (GLenum pname, const GLint *params);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:4:0:
/usr/include/GL/glew.h:16090:28: note: previous declaration of ‘PFNGLFRAGMENTLIGHTMODELIVSGIXPROC’ was here
 typedef void (GLAPIENTRY * PFNGLFRAGMENTLIGHTMODELIVSGIXPROC) (GLenum pname, GLint* params);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:7:0:
/usr/include/GL/glext.h:12079:25: error: conflicting types for ‘PFNGLGETFRAGMENTMATERIALFVSGIXPROC’
 typedef void (APIENTRYP PFNGLGETFRAGMENTMATERIALFVSGIXPROC) (GLenum face, GLenum pname, GLfloat *params);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:4:0:
/usr/include/GL/glew.h:16101:28: note: previous declaration of ‘PFNGLGETFRAGMENTMATERIALFVSGIXPROC’ was here
 typedef void (GLAPIENTRY * PFNGLGETFRAGMENTMATERIALFVSGIXPROC) (GLenum face, GLenum pname, const GLfloat* data);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:7:0:
/usr/include/GL/glext.h:12080:25: error: conflicting types for ‘PFNGLGETFRAGMENTMATERIALIVSGIXPROC’
 typedef void (APIENTRYP PFNGLGETFRAGMENTMATERIALIVSGIXPROC) (GLenum face, GLenum pname, GLint *params);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mujoco_py/mujoco_py/gl/eglshim.c:4:0:
/usr/include/GL/glew.h:16102:28: note: previous declaration of ‘PFNGLGETFRAGMENTMATERIALIVSGIXPROC’ was here
 typedef void (GLAPIENTRY * PFNGLGETFRAGMENTMATERIALIVSGIXPROC) (GLenum face, GLenum pname, const GLint* data);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Traceback (most recent call last):
  File "/usr/lib/python3.6/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/usr/lib/python3.6/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/usr/lib/python3.6/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/usr/lib/python3.6/distutils/spawn.py", line 159, in _spawn_posix
[... 77 lines cut ...]

**Expected behavior**
Build succeeds.

**Error Messages**
See above.

**Desktop (please complete the following information):**
 - OS: `Ubuntu 18.04`
 - Python Version: `3.6.5`
 - Mujoco Version: `1.50` and `2.00`
 - mujoco-py version:  `1.50.1.68` and `2.0.2.0`

**Environment**
 - output of: `echo $LD_LIBRARY_PATH`:
`/usr/local/nvidia/lib64:/root/.mujoco/mujoco200/bin::/usr/lib/nvidia-000`

 - output of: `echo $HOME`
`/root`

 - output of: `echo $USER`
``

**Additional context**
n/a

hartikainen, 2019-04-09:
I've narrowed down the issue to be the `libegl1-mesa-dev` apt package (or possibly some of it's dependency). I have a working docker image which brakes if I do `apt-get install libglew-dev`. I still don't know how to downgrade it back to the working version.

Here's the diff of `apt list --installed` for the working (`18.2.2`) vs. broken (`18.2.8`) installation:
```diff
-libegl1-mesa-dev/bionic-updates,now 18.2.8-0ubuntu0~18.04.2 amd64 [installed]
+libegl1-mesa-dev/bionic-updates,now 18.2.2-0ubuntu1~18.04.2 amd64 [installed]
```

hou-yz, 2019-06-22:
same problem here

hou-yz, 2019-06-23:
check this issue for fixes:
https://github.com/mupen64plus/mupen64plus-video-z64/pull/16

qxcv, 2019-07-21:
+1 for fix. In mujoco-py, I found it sufficient to simply remove `#include <GL/glext.h>` from `mujoco_py/gl/eglshim.c`.

Kelly510, 2021-05-22:
> +1 for fix. In mujoco-py, I found it sufficient to simply remove `#include <GL/glext.h>` from `mujoco_py/gl/eglshim.c`.

The eglshim.c file doesn't include <GL/glext.h> but I still have the problem. Does anyone know what headers else should be removed?

fokx, 2021-11-10:
The error info indicates conflicting types are defined both in `/usr/include/GL/glew.h` and `/usr/include/GL/glext.h`. So instead of 

>  remove `#include <GL/glext.h>` from `mujoco_py/gl/eglshim.c`

, which could introduce `unknown type name` error, it's preferred to comment out some lines in `/usr/include/GL/glext.h`. So L12306 to L12320 should be like:
//typedef void (APIENTRYP PFNGLFRAGMENTLIGHTFVSGIXPROC) (GLenum light, GLenum pname, const GLfloat *params);
typedef void (APIENTRYP PFNGLFRAGMENTLIGHTISGIXPROC) (GLenum light, GLenum pname, GLint param);
//typedef void (APIENTRYP PFNGLFRAGMENTLIGHTIVSGIXPROC) (GLenum light, GLenum pname, const GLint *params);
typedef void (APIENTRYP PFNGLFRAGMENTLIGHTMODELFSGIXPROC) (GLenum pname, GLfloat param);
//typedef void (APIENTRYP PFNGLFRAGMENTLIGHTMODELFVSGIXPROC) (GLenum pname, const GLfloat *params);
typedef void (APIENTRYP PFNGLFRAGMENTLIGHTMODELISGIXPROC) (GLenum pname, GLint param);
//typedef void (APIENTRYP PFNGLFRAGMENTLIGHTMODELIVSGIXPROC) (GLenum pname, const GLint *params);
typedef void (APIENTRYP PFNGLFRAGMENTMATERIALFSGIXPROC) (GLenum face, GLenum pname, GLfloat param);
typedef void (APIENTRYP PFNGLFRAGMENTMATERIALFVSGIXPROC) (GLenum face, GLenum pname, const GLfloat *params);
typedef void (APIENTRYP PFNGLFRAGMENTMATERIALISGIXPROC) (GLenum face, GLenum pname, GLint param);
typedef void (APIENTRYP PFNGLFRAGMENTMATERIALIVSGIXPROC) (GLenum face, GLenum pname, const GLint *params);
typedef void (APIENTRYP PFNGLGETFRAGMENTLIGHTFVSGIXPROC) (GLenum light, GLenum pname, GLfloat *params);
typedef void (APIENTRYP PFNGLGETFRAGMENTLIGHTIVSGIXPROC) (GLenum light, GLenum pname, GLint *params);
//typedef void (APIENTRYP PFNGLGETFRAGMENTMATERIALFVSGIXPROC) (GLenum face, GLenum pname, GLfloat *params);
//typedef void (APIENTRYP PFNGLGETFRAGMENTMATERIALIVSGIXPROC) (GLenum face, GLenum pname, GLint *params);